lvgl/lvgl #6104: Add back CONFIG_LV_COLOR_16_SWAP

XuNeo, 2024-04-22:
### Introduce the problem

While migrating all existing project using lvgl on Nuttx, https://github.com/apache/nuttx/pull/11780, many configurations have enabled CONFIG_LV_COLOR_16_SWAP. It may be a common issue others will face too.

```bash
nuttx/boards/risc-v/esp32c3-legacy/esp32c3-devkit/configs/lvgl/defconfig:38:CONFIG_LV_COLOR_16_SWAP=y
nuttx/boards/xtensa/esp32/esp32-2432S028/configs/lvgl/defconfig:39:CONFIG_LV_COLOR_16_SWAP=y
nuttx/boards/xtensa/esp32/esp32-wrover-kit/configs/lvgl/defconfig:44:CONFIG_LV_COLOR_16_SWAP=y
nuttx/boards/xtensa/esp32s2/esp32s2-kaluga-1/configs/lvgl_ili9341/defconfig:42:CONFIG_LV_COLOR_16_SWAP=y
nuttx/boards/xtensa/esp32s2/esp32s2-kaluga-1/configs/lvgl_st7789/defconfig:44:CONFIG_LV_COLOR_16_SWAP=y
nuttx/boards/xtensa/esp32s3/esp32s3-box/configs/lvgl-3/defconfig:56:CONFIG_LV_COLOR_16_SWAP=y
nuttx/boards/xtensa/esp32s3/esp32s3-box/configs/lvgl/defconfig:59:CONFIG_LV_COLOR_16_SWAP=y
nuttx/boards/xtensa/esp32/ttgo_t_display_esp32/configs/lvgl_lcd/defconfig:45:CONFIG_LV_COLOR_16_SWAP=y
nuttx/defconfig:39:CONFIG_LV_COLOR_16_SWAP=y
```

### Proposal

Add equivalent configuration option `CONFIG_LV_COLOR_16_SW_SWAP` that does the swap by software just before `flush_cb` by calling `lv_draw_sw_rgb565_swap`.

Update existing configuration to use `CONFIG_LV_COLOR_16_SW_SWAP` instead.

XuNeo, 2024-04-24:
Mainly for backward compatibility. Without `CONFIG_LV_COLOR_16_SW_SWAP `, those listed configurations will be break in lvgl v9 and it's tough for me to upgrade all LCD drivers to support hardware swap.

kisvegabor, 2024-04-25:
Makes sense, can you prepare a PR?

FASTSHIFT, 2024-04-28:
If the LCD driver does not support the endian setting, the SPI can be switched to Bit Order set to MSB FIRST+16bit mode when sending the image buffer with DMA, and then switched to 8bit mode after sending, so that the code is simple and efficient. 
I believe most MCUs can support this mode.

See:
[image]

[image]

XuNeo, 2024-04-29:
Nice work.

Calling `lv_draw_sw_rgb565_swap` also has limitation. It performs byte order swap in place, thus needs to swap back before drawing next frame.

kisvegabor, 2024-04-29:
> It performs byte order swap in place, thus needs to swap back before drawing next frame.

Why? The buffer into which the UI drawn is dropped after sending the image. Or do you use LVGL in direct mode?

XuNeo, 2024-04-29:
Yes, we have to consider direct mode too.

kisvegabor, 2024-04-29:
In this case we should swap the bytes only on a given area. Adding `lv_draw_sw_rgb565_swap_area` can be useful.

PR #6225 (2024-05-24): feat(refr): add backward compatibility for LV_COLOR_16_SWAP
fixes #6104

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py)  [... cut]
closed 2024-05-24 by kisvegabor