lvgl/lvgl #6108: Documentation examples do not load in Chrome OSX when selected branch is master

plueschpruem, 2024-04-22:
### LVGL version

any

### What happened?

When I have the docu set to "master (latest)", I get this in the console:
**Subresource requests whose URLs contain embedded credentials (e.g. `https://user:pass@host/`) are blocked. See https://www.chromestatus.com/feature/5669008342777856 for more details.**

Seems like Chrome doesn't like the format of the iFrame tags…
`rc="//**.* @file lvgl.h/_static/built_lv_examples/index.html?example=lv_example_get_started_1&amp;w=320&amp;h=240"`

Selecting "v9.1" shows the examples.
Those tags have a different format:
`src="/9.1/_static/built_lv_examples/index.html?example=lv_example_event_1&w=320&h=240"`

kisvegabor, 2024-04-25:
Should be fixed now!
https://docs.lvgl.io/master/examples.html

plueschpruem, 2024-04-25:
At the moment the whole page is 404 for me :-D

kisvegabor, 2024-04-29:
Now all good :slightly_smiling_face:

closed 2024-05-14 by kisvegabor