lvgl/lvgl #6111: https://docs.lvgl.io/8.3/ link leads to 404

Russtopia, 2024-04-22:
### LVGL version

v9.1.0

### What happened?

Sometime since my last visit to the official online LVGL docs at https://docs.lvgl.io/, 

* v8.0 docs have completely disappeared from the available versions to browse online;
* v8.3 selection leads to a 404 page.

What is the policy for retaining online access from the official docs pages? Should I self-host docs if I want to browse older API docs online?

### How to reproduce?

1. Visit https://docs.lvgl.io/
2. Observer v8.0 is no longer browseable.
3. Access version dropdown in NW corner of page, select v8.3 (https://docs.lvgl.io/8.3/)
4. Observe 404 error.

kisvegabor, 2024-04-23:
Thank for opening this issue. I'm working on it.

embeddedt, 2024-04-24:
> for master the src of the example's iframes is still set incorrectly:

Should be fixed with https://github.com/lvgl/lvgl/pull/6122.

Note: I had some difficulty getting the docs to build on my system. I cloned the repository and ran the exact same commands as the CI workflow, but I get the following error:

```
... doxygen log above ...
Reading Doxygen output
Traceback (most recent call last):
  File "/home/tt/work/lvgl/docs/build.py", line 147, in <module>
    doc_builder.run(
  File "/home/tt/work/lvgl/docs/doc_builder.py", line 663, in run
    globals()[compound.attrib['kind'].upper()](
  File "/home/tt/work/lvgl/docs/doc_builder.py", line 299, in __init__
    member.attrib['name'] = member[0].text.strip()
                            ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'strip'
```

I hacked around this for testing the version fix by commenting out the call to `doc_builder.run()`, but I'd like to fix it properly. Evidently CI is building it without problems so I'm not sure what the difference is on my local machine (possibly a newer version of Doxygen, since I use Arch Linux). Is there some documentation on what `doc_builder.py` is for?

kdschlosser, 2024-04-25:
OK I am coming up with errors in these files.

lv_draw_sw_mask.h
lv_symbol_def.h
stb_truetype_htcw.h
lv_scale.h
lv_textarea.h

I have to do some looking at the documentation in those files to see what is causing the issue.

I can tell you that there is a very large number of problems in the C documentation all over LVGL. incorrect parameter names for functions, missing parameter names. all kinds of stuff.

There are issues with the heading lines used in the doc files (`.rst`) 

<details>
  <summary>Click for list of errors</summary>

```
WARNING: lvgl/src/draw/lv_draw_triangle.h:53: warning: The following parameter of lv_draw_triangle(lv_layer_t *layer, const lv_draw_triangle_dsc_t *draw_dsc) is not documented:
  parameter 'draw_dsc'
WARNING: lvgl/src/draw/lv_draw_triangle.h:40: warning: The following parameter of lv_draw_triangle_dsc_init(lv_draw_triangle_dsc_t *draw_dsc) is not documented:
  parameter 'draw_dsc'
WARNING: lvgl/src/draw/sw/lv_draw_sw.h:94: warning: The following parameter of lv_draw_sw_image(lv_draw_unit_t *draw_unit, const lv_draw_image_dsc_t *draw_dsc, const lv_area_t *coords) is not documented:
  parameter 'draw_dsc'
WARNING: lvgl/src/draw/sw/lv_draw_sw.h:126: warning: The following parameter of lv_draw_sw_layer(lv_draw_unit_t *draw_unit, const lv_draw_image_dsc_t *draw_dsc, const lv_area_t *coords) is not documented:
  parameter 'draw_dsc'
WARNING: lvgl/src/draw/sw/lv_draw_sw.h:175: warning: The following parameter of lv_draw_sw_rgb565_swap(void *buf, uint32_t buf_size_px) is not documented:
  parameter 'buf'
WARNING: lvgl/src/draw/sw/lv_draw_sw.h:149: warning: The following parameter of lv_draw_sw_transform(lv_draw_unit_t *draw_unit, const lv_area_t *dest_area, const void *src_buf, int32_t src_w, int32_t src_h, int32_t src_stride, const lv_draw_image_dsc_t *draw_dsc, const lv_draw_image_sup_t *sup, lv_color_format_t cf, void *dest_buf) is not documented:
  parameter 'draw_dsc'
WARNING: lvgl/src/draw/sw/lv_draw_sw_gradient.h:42: warning: The following parameters of lv_gradient_color_calculate(const lv_grad_dsc_t *dsc, int32_t range, int32_t frac, lv_grad_color_t *color_out, lv_opa_t *opa_out) are not documented:
  parameter 'color_out'
  parameter 'opa_out'
WARNING: lvgl/src/draw/sw/lv_draw_sw_mask.h:241: warning: The following parameter of lv_draw_sw_mask_line_angle_init(lv_draw_sw_mask_line_param_t *param, int32_t p1x, int32_t py, int16_t angle, lv_draw_sw_mask_line_side_t side) is not documented:
  parameter 'p1x'
WARNING: lvgl/src/font/lv_binfont_loader.h:30: warning: The following parameter of lv_binfont_create(const char *font_name) is not documented:
  parameter 'font_name'
WARNING: lvgl/src/misc/lv_anim.h:384: warning: The following parameters of lv_anim_set_bezier3_param(lv_anim_t *a, int16_t x1, int16_t y1, int16_t x2, int16_t y2) are not documented:
  parameter 'x2'
  parameter 'y2'
WARNING: lvgl/src/misc/lv_anim.h:315: warning: The following parameter of lv_anim_set_playback_duration(lv_anim_t *a, uint32_t duration) is not documented:
  parameter 'duration'
WARNING: lvgl/src/misc/lv_color.h:461: warning: The following parameter of lv_color16_luminance(const lv_color16_t c) is not documented:
  parameter 'c'
WARNING: lvgl/src/misc/lv_color.h:471: warning: The following parameter of lv_color24_luminance(const uint8_t *c) is not documented:
  parameter 'c'
WARNING: lvgl/src/misc/lv_color.h:481: warning: The following parameter of lv_color32_luminance(lv_color32_t c) is not documented:
  parameter 'c'
WARNING: lvgl/src/misc/lv_color.h:209: warning: The following parameter of lv_color_format_get_bpp(lv_color_format_t cf) is not documented:
  parameter 'cf'
WARNING: lvgl/src/misc/lv_color.h:216: warning: The following parameter of lv_color_format_get_size(lv_color_format_t cf) is not documented:
  parameter 'cf'
WARNING: lvgl/src/misc/lv_color.h:451: warning: The following parameter of lv_color_luminance(lv_color_t c) is not documented:
  parameter 'c'
WARNING: lvgl/src/misc/lv_color_op.h:85: warning: The following parameter of lv_color_brightness(lv_color_t c) is not documented:
  parameter 'c'
WARNING: lvgl/src/others/observer/lv_observer.h:312: warning: The following parameter of lv_obj_bind_state_if_eq(lv_obj_t *obj, lv_subject_t *subject, lv_state_t state, int32_t ref_value) is not documented:
  parameter 'state'
WARNING: lvgl/src/others/observer/lv_observer.h:322: warning: The following parameter of lv_obj_bind_state_if_not_eq(lv_obj_t *obj, lv_subject_t *subject, lv_state_t state, int32_t ref_value) is not documented:
  parameter 'state'
WARNING: lvgl/src/others/observer/lv_observer.h:242: warning: The following parameter of lv_subject_add_observer_with_target(lv_subject_t *subject, lv_observer_cb_t cb, void *target, void *user_data) is not documented:
  parameter 'cb'
WARNING: lvgl/src/others/observer/lv_observer.h:177: warning: The following parameter of lv_subject_init_color(lv_subject_t *subject, lv_color_t color) is not documented:
  parameter 'color'
WARNING: lvgl/src/others/observer/lv_observer.h:259: warning: The following parameter of lv_subject_remove_all_obj(lv_subject_t *subject, lv_obj_t *obj) is not documented:
  parameter 'subject'
WARNING: lvgl/src/others/observer/lv_observer.h:184: warning: The following parameter of lv_subject_set_color(lv_subject_t *subject, lv_color_t color) is not documented:
  parameter 'color'
[... cut ...]
      --------^
  If the function has a return type:
    Error in declarator or parameters-and-qualifiers
    If pointer to member declarator:
      Invalid C++ declaration: Expected '::' in pointer to member (function). [error at 13]
        int32_t diff = lv_event_get_rotary_diff(e)
        -------------^
    If declarator-id:
      Invalid C++ declaration: Expecting "(" in parameters-and-qualifiers. [error at 13]
        int32_t diff = lv_event_get_rotary_diff(e)
        -------------^
```

</details>

kdschlosser, 2024-04-25:
I can get the docs to build without fixing the errors, but the documentation will not be correct and things will not link correctly.

Up to you to decide to correct the errors or just patch it to get them to build.

kdschlosser, 2024-04-25:
What is stopping the build is caused by something in the LVGL source files and not the build system. I am able to alter the build system so it will build but that is not going to correct the actual problem, it will only end up skipping over the problem spots. I wrote the build system in a manner that would cause the build to fail if something is not done properly. I did that intentionally so things can get fixed.

I will do whatever you want me to do.

kisvegabor, 2024-04-25:
I think we should make it compile (although it compiles in CI) and fix the docs error step by step.

kdschlosser, 2024-04-25:
The PR that was done for correcting the version needs to be reversed as it is the wrong thing to do. I made a comment about it in that PR

kdschlosser, 2024-04-25:
The only thing I want to point out about altering the build system so it doesn't fail is that you will need to go and specifically check the output of the CI to see if there are errors and then go back and fix them.

I do not know why the CI is passing the build That is something that has to be looked into as to why it is not looking at the exit codes to determine that there is a failure that has occurred.

Russtopia, 2024-05-10:
Hi, thanks for working on this.I browsed a few versions online now(8.1, 8.3, 8.4) and the docs look like they're up again.

kdschlosser, 2024-05-10:
yeah you can close this issue

closed 2024-05-10 by Russtopia