lvgl/lvgl #6116: snapshot  question

chenyinen, 2024-04-23:
### LVGL version

v8.3

### What happened?

lv_example_snapshot_1.c
In this file, each capture will release the cache of the previous capture, but if LVGL enables cache, will there be a memory problem, and whether it should be called here lv_img_cache_invalidate_src when calling lv_snapshot_free

### How to reproduce?

static void event_cb(lv_event_t * e)
{
    lv_obj_t * snapshot_obj = lv_event_get_user_data(e);
    lv_obj_t * img = lv_event_get_target(e);

    if(snapshot_obj) {
        lv_img_dsc_t * snapshot = (void *)lv_img_get_src(snapshot_obj);
        if(snapshot) {
            lv_snapshot_free(snapshot);
        }

        /*Update the snapshot, we know parent of object is the container.*/
        snapshot = lv_snapshot_take(img->parent, LV_IMG_CF_TRUE_COLOR_ALPHA);
        if(snapshot == NULL)
            return;
        lv_img_set_src(snapshot_obj, snapshot);
    }
}

kisvegabor, 2024-04-28:
Just to confirm: do you mean each capture should release the cache of the previous capture, but it doesn't happen, therefore the image is not updated?

If so, you are right! I think the cache should be invalidated  in `lv_snapshot_free`. Do you mean the same?

kisvegabor, 2024-05-02:
Great, could you send a Pull request with the fix you have proposed?

kisvegabor, 2024-05-17:
Not stale

closed 2024-06-20 by kisvegabor