lvgl/lvgl #6118: Copying buffers when using DIORECT and double buffering.

kdschlosser, 2024-04-23:
### Introduce the problem

Here is the code that handles the updating of the buffer that was just flushed.

```c
static void refr_sync_areas(void)
{
    /*Do not sync if not direct or double buffered*/
    if(disp_refr->render_mode != LV_DISPLAY_RENDER_MODE_DIRECT) return;

    /*Do not sync if not double buffered*/
    if(!lv_display_is_double_buffered(disp_refr)) return;

    /*Do not sync if no sync areas*/
    if(_lv_ll_is_empty(&disp_refr->sync_areas)) return;

    LV_PROFILER_BEGIN;
    /*With double buffered direct mode synchronize the rendered areas to the other buffer*/
    /*We need to wait for ready here to not mess up the active screen*/
    wait_for_flushing(disp_refr);

    /*The buffers are already swapped.
     *So the active buffer is the off screen buffer where LVGL will render*/
    lv_draw_buf_t * off_screen = disp_refr->buf_act;
    lv_draw_buf_t * on_screen = disp_refr->buf_act == disp_refr->buf_1 ? disp_refr->buf_2 : disp_refr->buf_1;

    uint32_t hor_res = lv_display_get_horizontal_resolution(disp_refr);
    uint32_t ver_res = lv_display_get_vertical_resolution(disp_refr);

    /*Iterate through invalidated areas to see if sync area should be copied*/
    uint16_t i;
    int8_t j;
    lv_area_t res[4] = {0};
    int8_t res_c;
    lv_area_t * sync_area, * new_area, * next_area;
    for(i = 0; i < disp_refr->inv_p; i++) {
        /*Skip joined areas*/
        if(disp_refr->inv_area_joined[i]) continue;

        /*Iterate over sync areas*/
        sync_area = _lv_ll_get_head(&disp_refr->sync_areas);
        while(sync_area != NULL) {
            /*Get next sync area*/
            next_area = _lv_ll_get_next(&disp_refr->sync_areas, sync_area);

            /*Remove intersect of redraw area from sync area and get remaining areas*/
            res_c = _lv_area_diff(res, sync_area, &disp_refr->inv_areas[i]);

            /*New sub areas created after removing intersect*/
            if(res_c != -1) {
                /*Replace old sync area with new areas*/
                for(j = 0; j < res_c; j++) {
                    new_area = _lv_ll_ins_prev(&disp_refr->sync_areas, sync_area);
                    *new_area = res[j];
                }
                _lv_ll_remove(&disp_refr->sync_areas, sync_area);
                lv_free(sync_area);
            }

            /*Move on to next sync area*/
            sync_area = next_area;
        }
    }

    lv_area_t disp_area = {0, 0, (int32_t)hor_res - 1, (int32_t)ver_res - 1};
    /*Copy sync areas (if any remaining)*/
    for(sync_area = _lv_ll_get_head(&disp_refr->sync_areas); sync_area != NULL;
        sync_area = _lv_ll_get_next(&disp_refr->sync_areas, sync_area)) {
        /**
         * @todo Resize SDL window will trigger crash because of sync_area is larger than disp_area
         */
        _lv_area_intersect(sync_area, sync_area, &disp_area);
        lv_draw_buf_copy(off_screen, sync_area, on_screen, sync_area);
    }

    /*Clear sync areas*/
    _lv_ll_clear(&disp_refr->sync_areas);
    LV_PROFILER_END;
}
```

### Proposal

To me it would make more sense to let the buffer that is currently being flushed and have it read to fill the other buffer.

Currently how it is done is like this,

buf 1 flushing to screen, this is done using DMA so the processor doesn't get tied up
LVGL fills buf 2
program stalls until buf 1 has finished being dumped top the display
invalidated areas are copied from buf 2 to buf 1
buf 2 gets flushed to the display

If we change the orientation of when things are done it would lead to a more efficient system

buf 1 flushing to display
as buf 1 is being written to the display copy the invalid areas from the buffer to buf 2
buf 2 gets updated by LVGL
hold program until buf 1 is fully flushed.
write buf 2 to the display

This would be far better management of time if the display size is large, like 800 x 480. The amount of time it takes to flush the buffer can exceed the length of time it takes for LVGL to render new data. If that is the case now there is going to be a long pause and then copying of invalidated areas after that. The invalidated areas could have been copied while the buffer was being flushed which would reduce the time by how long it takes to copy the invalidated areas.

We cannot change the data in the buffer while it is being flushed, but we sure can read the data to copy invalidated areas.

It seems like making a simple change to when things are being done could make a decent sized improvement when dealing with large displays.

The other question, would it not be faster to simply do a memcpy on the entire buffer VS iterating over areas that are invalidated and then iterating over each line of the invalidated area to copy one line at a time of the invalidated area. I would imagine this would be dependent on how much is being copied. If 1/3 the buffer needs to be updated it might be faster to just memcpy the entire buffer. It's a whole lot less operations that get done when using memcpy VS invalidated areas.

kdschlosser, 2024-04-25:
I am testing this using SDL locally but the same thing is being seen wen running an actual MCU with a display that has  and RGB connection

I am not able to wrap my head around the order in which things are being done. It looks like the copying of the buffer is being done before the invalidated areas are rendered but the notes in the file say otherwise.

Are the notes incorrect? 

I feel like there is a stall in the code that is causing a wait until the buffer gets emptied before doing anything. Even when removing that one wait that is in the code I still end up with a 15 millisecond hang when calling lv_task_handler. That seems like a very large amount of time for rendering 2 sliders. While I know the pixel count is high at 800 x 480 having that kind of a stall seems to be pretty excessive. This is also running on a PC and not a micro controller so the stall would be even longer on a microcontroller.

kdschlosser, 2024-04-25:
This is what a user is running as a test...

```python

scrn = lv.screen_active()
lslider = lv.slider(scrn)
lslider.set_style_size(225, 480, lv.PART.MAIN)
lslider.align(lv.ALIGN.TOP_LEFT, 0, 0)
rslider = lv.slider(scrn)
rslider.set_style_size(225, 480, lv.PART.MAIN)
rslider.align(lv.ALIGN.TOP_RIGHT, 0, 0)

slider_1_inc = 1
slider_2_inc = -1
slider_1_value = 0
slider_2_value = 100

while True:
    time.sleep_ms(10)
    lslider.set_value(slider_1_value, lv.ANIM.OFF)

    slider_1_value += slider_1_inc
    if slider_1_value in (0, 100):
        slider_1_inc = -slider_1_inc

    rslider.set_value(slider_2_value, lv.ANIM.OFF)

    slider_2_value += slider_2_inc
    if slider_2_value in (0, 100):
        slider_2_inc = -slider_2_inc

    start_time = time.ticks_ms()
    lv.refr_now(lv.display_get_default())
    stop_time = time.ticks_ms()
    print('refr time:', time.tick_diff(stop_time, start_time))
    print()
    lv.tick_inc(10)
```

This test is only testing how long it takes for lv_refr_now to run after updating the 2 slider widgets

This is what I am getting for output
```
refr time: 185
refr time: 186
refr time: 186
```

Now change the code so a refresh is done after each widget has it's value changed

```python

scrn = lv.screen_active()
lslider = lv.slider(scrn)
lslider.set_style_size(225, 480, lv.PART.MAIN)
lslider.align(lv.ALIGN.TOP_LEFT, 0, 0)
rslider = lv.slider(scrn)
rslider.set_style_size(225, 480, lv.PART.MAIN)
rslider.align(lv.ALIGN.TOP_RIGHT, 0, 0)

slider_1_inc = 1
slider_2_inc = -1
slider_1_value = 0
slider_2_value = 100

while True:
    time.sleep_ms(10)
    lslider.set_value(slider_1_value, lv.ANIM.OFF)
    lv.refr_now(lv.display_get_default())

    slider_1_value += slider_1_inc
    if slider_1_value in (0, 100):
        slider_1_inc = -slider_1_inc

    rslider.set_value(slider_2_value, lv.ANIM.OFF)

    slider_2_value += slider_2_inc
    if slider_2_value in (0, 100):
        slider_2_inc = -slider_2_inc

    start_time = time.ticks_ms()
    lv.refr_now(lv.display_get_default())
    stop_time = time.ticks_ms()
    print('refr time:', time.tick_diff(stop_time, start_time))
    print()
    lv.tick_inc(10)

```

so now I am only doing a refresh of one of the sliders and this is the output
```
refr time: 98
refr time: 98
refr time: 98
refr time: 98
```

The flickering and tearing is also reduced by 95%

98 milliseconds seems like a really long time for the refresh to run for. This is an RGB565 display that is 800 x 480 connected using an RGB bus with double buffering.

kdschlosser, 2024-04-25:
forgot to mention that is running on an ESP32-S3

closed 2024-05-22 by lvgl-bot

kisvegabor, 2024-05-22:
#6120 is merged which remeves an unnecessary blocking wait.