lvgl/lvgl #6119: corners of bg_image are not clipped in objects

turoksama, 2024-04-24:
### LVGL version

v9.1

### What happened?

Hi there, here we have a situation as mentioned above. I am not sure it's a bug whether or not.

[image]

### How to reproduce?

void obj_bg_clip_test(void){
	lv_obj_t *obj = lv_obj_create(lv_screen_active());
	lv_obj_set_size( obj, 480, 272);
	lv_obj_align(obj, LV_ALIGN_CENTER, 0, 0);
	lv_obj_set_style_bg_image_src(obj, &bg16, 0);
	lv_obj_set_style_bg_image_tiled(obj, 1, 0);
	lv_obj_set_style_clip_corner(obj, 1, 0);
	lv_obj_set_style_radius(obj, 272/2, 0);
}

kisvegabor, 2024-04-24:
I could reproduce it, but still not sure how to fix it the best way. Probably the image rendering should be extended at least with a radius parameter. 

@FASTSHIFT, @nicusorcitu, @anaGrad  can you handle it VG-Lite too?

nicusorcitu, 2024-04-24:
@cristian-stoica

nicusorcitu, 2024-04-24:
I believe the VGLite code already covers that if there is a a radius descriptor.
We observed that in our unit tests as well. Circle also had no rounded background image because of missing the radius descriptor.

FASTSHIFT, 2024-04-24:
My implementation: https://github.com/lvgl/lvgl/pull/6121

turoksama, 2024-04-24:
Still the same effect in bg_image with tiled enabled. @FASTSHIFT 

> void obj_bg_clip_test(void){
lv_obj_t *obj = lv_obj_create(lv_screen_active());
lv_obj_set_size( obj, 480, 272);
lv_obj_align(obj, LV_ALIGN_CENTER, 0, 0);
lv_obj_set_style_bg_image_src(obj, &bg16, 0);
lv_obj_set_style_bg_image_tiled(obj, 1, 0);
lv_obj_set_style_clip_corner(obj, 1, 0);
lv_obj_set_style_radius(obj, 272/2, 0);
}

kisvegabor, 2024-04-28:
As I mentioned it [here](https://github.com/lvgl/lvgl/pull/6121#discussion_r1582083082), the `radius` should be applied on any images not only on layer. 

For tiled images we need not only the radius but the larger area on which the radius shall be applied. Fortunately we have [`image_draw_dsc.image_area`](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_image.h#L62-L67) which gives the full image area and radius should be applied on this area.

nicusorcitu, 2024-05-13:
@anaGrad @cristian-stoica can you please verify if this is fixed in our rounded rectangle (or circle) unit test with background image?
Thanks.

kisvegabor, 2024-05-16:
Theoretical question: If the image is rotated the radius mask shall be rotated too. That is the rounded image is rotated and not the rotated image is masked with a normal radius mask. I think this makes the most sense but it's quite complicated to implement in SW rendering as the mask needs to be rotated too.

anaGrad, 2024-05-22:
I tested the PR for this ticket and I have a mention about the implementation. 
I noticed that the rounded corner image works only if done by creating an image widget with lv_image_create(). On our tests, we have a rectangle with background image, and so the radius doesn't apply to it. 
The simplest way I found it possible to pass the radius to the image in that case was by giving the image radius the value from the rectangle in lv_draw_rect. 
Please take a look at the patch below and tell me if there is something else I should take into account or if it's ok for it to be also included in the PR.

[image_radius.patch](https://github.com/lvgl/lvgl/files/15402522/image_radius.patch)

kisvegabor, 2024-05-27:
@anaGrad 
True, bg_image drawing should know the radius too.

@nicusorcitu 
At the first pass we can accept it as a limitation of the SW render and support it only for non rotated images.

closed 2024-08-24 by lvgl-bot