lvgl/lvgl #6131: Add support for CSS-like Text Leading-Trim

GIldons, 2024-04-26:
### Problem to solve

Currently, LVGL text alignment within a container is based on the worst-case scenario height of a font as generated: The font height is calculated as the height between the descender and ascender lines, and its total height depends on the characters used to generate the font. For example, all other configs being equal, the generated font height will be different if generated for A-Z vs A-Z, a-z.

As a result, having, for a worst-case example, a label and an image added to a button will result in the label looking off-center and higher than the image if the font used for the label contains a non-zero descender height. This effect, however, can be noticed with just a label and is more noticeable with bigger font sizes.

This article has a quick description of the solution:
https://css-tricks.com/leading-trim-the-future-of-digital-typesetting/

This article has good examples of the problem
https://medium.com/microsoft-design/leading-trim-the-future-of-digital-typesetting-d082d84b202

### Success criteria

Being able to archive proper vertical text alignment regardless of the character/font set used.

### Solution outline

Add enough information to the generated fonts to allow trimming the text box for purposes of text alignment.

### Rabbit holes

This might require a dedicated set of text_align functions, however, I think this might be accomplished using the generic lv_object_align functions.

### Testing

Visual inspection of new font alignment including different sets of characters and fonts.

### Teaching

Updated documentation code snippets.

kisvegabor, 2024-04-29:
Hey,

I'm thinking about translating the label with its baseline:
```c
  lv_obj_t * cont = lv_obj_create(lv_screen_active());
  lv_obj_set_size(cont, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
  lv_obj_set_flex_flow(cont, LV_FLEX_FLOW_ROW);
  lv_obj_set_style_flex_cross_place(cont, LV_FLEX_ALIGN_END, 0);

  LV_IMAGE_DECLARE(emoji_F617);
  lv_obj_t * icon = lv_image_create(cont);
  lv_obj_set_pos(icon, 100, 100);
  lv_image_set_src(icon, &emoji_F617);

  lv_obj_t * label = lv_label_create(cont);
  lv_label_set_text(label, "Agy");
  lv_obj_set_style_text_font(label, &lv_font_montserrat_48, 0);
  lv_obj_set_pos(icon, 100, 100);
```
[image]

Adding
```c
  lv_obj_set_style_translate_y(label, lv_font_montserrat_48.base_line, 0); /*offset by 9*/
```
[image]

What do you think about it?

GIldons, 2024-04-29:
I think this would address the initial request we had in the forum topic, however, this would not play well with general alignment.

If instead of calling `lv_obj_set_pos`,  we call `lv_obj_set_align` on both the icon and the label, and to the result of that we translate the label by `base_line`, we are still going to be off alignment, as we are not taking into account the cap height.

With that said, I do think that what you proposed here has merit. What would be the strategy for this? Incorporate the translate call to a `lv_label_set_pos` kind of function?

kisvegabor, 2024-05-02:
You are right about that what I suggested solves only the bottom alignment, but doesn't solve the center and top alignment. 

I'm thinking about how can we support the options supported by CSS. As the [docs says](https://developer.mozilla.org/en-US/docs/Web/CSS/vertical-align):
> The vertical-align CSS property sets vertical alignment of an inline, inline-block or table-cell box.

However we don't have an `inline` layout. In LVGL position can be set either:
- manually: a different sizing model as you proposed can work to make the alignment simpler. I don't know how to make the widgets change their default positioning (there is no such a feature now).  For top alignment the cap height info would be required which is not stored now. 
- by flexbox and grid: In CSS there is a [baseline alignment](https://developer.mozilla.org/en-US/docs/Web/CSS/align-items) and we can support it in LVGL too.

In Figma it works like [this](https://www.youtube.com/watch?v=clWWBbxbRrM). It's basically a flexbox layout.  

It seems to me the low hanging fruit is to make it work in flex box, and show examples for manual alignment.

What do you think?

kisvegabor, 2024-05-09:
I just could connect this issue with another one. When an image is zoomed or rotated, the untransformed image is positioned. However, in some cases it would be nice to align the transformed image. To do this we need to store the transformed coordinates of the widget too and use it in the alignments. This way as the image is zoomed it will push away its neighbors. 

It's very similar to what you have recommended for the label: have a different area for the label which is used in the alignments. 

It seems a like a complex development as it requires updating all the positioning and layout algorithms. It's interesting indeed, but I would like to close some running feature developments before jumping into it .

This is issue is slightly related to https://github.com/lvgl/lvgl/issues/5938

kisvegabor, 2024-09-05:
I've created the roadmap for v9.3 and I couldn't add this feature now. :slightly_frowning_face: 

Let's see if it comes up again. If more people want is it will be easier to prioritize it too.

Viatorus, 2025-05-13:
I am very interested in this function. Maybe for 9.4?

AndreCostaaa, 2025-12-04:
Thank you for the feedback @aafloy !

closed 2026-04-20 by AndreCostaaa