lvgl/lvgl #6137: Keyboard keys are colored differently

blackchen-1, 2024-04-28:
### Introduce the problem

Can you make the keyboard keys uniform in color? Now some keys are white and some are gray
[image]

kisvegabor, 2024-04-29:
By default the control keys are grey. It's inspired by smartphone keyboards. Like this:
[image]

You can remove the grey "state" by adding this:
```c
lv_buttonmatrix_clear_button_ctrl_all(kb, LV_BUTTONMATRIX_CTRL_CHECKED);
```

blackchen-1, 2024-05-08:
If you switch modes, the color difference changes back. How to solve this problem. Make a judgment to switch modes. Didn't toggle to add lv_buttonmatrix_clear_button_ctrl_all at once?

kisvegabor, 2024-05-13:
Ah, that's true! In this case please add this: 
```c
lv_obj_set_style_bg_color(kb, lv_color_white(), LV_PART_ITEMS | LV_STATE_CHECKED);
```

closed 2024-06-11 by lvgl-bot