lvgl/lvgl #6138: touchscreen no work with sdl + drm

Coloryr, 2024-04-28:
### LVGL version

v9.1.0

### What happened?

touchscreen no work runing in sdl + drm
```
/*==================
 * DEVICES
 *==================*/
/*Use SDL to open window on PC and handle mouse and keyboard*/
#define LV_USE_SDL              1
#if LV_USE_SDL
    #define LV_SDL_INCLUDE_PATH     <SDL2/SDL.h>
    #define LV_SDL_RENDER_MODE      LV_DISPLAY_RENDER_MODE_DIRECT   /*LV_DISPLAY_RENDER_MODE_DIRECT is recommended for best performance*/
    #define LV_SDL_BUF_COUNT        1    /*1 or 2*/
    #define LV_SDL_FULLSCREEN       0    /*1: Make the window full screen by default*/
    #define LV_SDL_DIRECT_EXIT      1    /*1: Exit the application when all SDL windows are closed*/
    #define LV_SDL_MOUSEWHEEL_MODE  LV_SDL_MOUSEWHEEL_MODE_ENCODER  /*LV_SDL_MOUSEWHEEL_MODE_ENCODER/CROWN*/
#endif
```

I touch the screen, it can go to https://github.com/lvgl/lvgl/blob/cac17b58071edec8b7a81ab727b57045bb440d6c/src/drivers/sdl/lv_sdl_mouse.c#L120
but, `indev` and `disp` always is `null` in https://github.com/lvgl/lvgl/blob/cac17b58071edec8b7a81ab727b57045bb440d6c/src/drivers/sdl/lv_sdl_mouse.c#L130C5-L141C30 from debug.
I sew `win_id` is 0, but no window with id 0 exists.

### How to reproduce?

```
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>

#include "lvgl.h"

#include "demos/lv_demos.h"

int main(int, char **)
{
    lv_init();

    lv_display_t *display = lv_sdl_window_create(640, 480);
    lv_indev_t *mouse = lv_sdl_mouse_create();
    lv_indev_t *mousewheel = lv_sdl_mousewheel_create();
    lv_indev_t *keyboard = lv_sdl_keyboard_create();

    lv_demo_widgets();

    for (;;)
    {
        uint32_t time = lv_timer_handler();
        usleep(time * 1000);
    }
}
```

kisvegabor, 2024-04-29:
I've checked in [SDL's docs](https://wiki.libsdl.org/SDL2/SDL_TouchFingerEvent) that for `SDL_FINGER*` events really `event->tfinger` shall be used. If SDL doesn't set the Window ID we cannot do much. 

Can you ask it from the SDL developers?

Coloryr, 2024-04-30:
I'm use is in DRM not X11, maybe these two are different.

Coloryr, 2024-04-30:
```
/*Find a suitable indev*/
    lv_indev_t * indev = lv_indev_get_next(NULL);
    while(indev) {
        if(disp != NULL) {
            if(lv_indev_get_display(indev) != disp) {
                break;
            }
        }
        if(lv_indev_get_type(indev) == LV_INDEV_TYPE_POINTER) {
            break;
        }
        indev = lv_indev_get_next(indev);
    }
```
I have modified the `lv_sdl_mouse.c` code, that it can work properly.

kisvegabor, 2024-05-02:
> I have modified the lv_sdl_mouse.c code, that it can work properly.

What is the difference? It seems to your code does the same thing, just the `if` is split into 2 parts. Am I missing something?

Coloryr, 2024-05-06:
In drm, `disp` is null, but lv_indev_get_display(indev) is not null.

kisvegabor, 2024-05-07:
This way always the first POINTER indev will be found. Which is better nothing, but worth mentioning. 

I've opened https://github.com/lvgl/lvgl/pull/6194 to fix it.

Coloryr, 2024-05-08:
I have tested it, this patch is work.

kisvegabor, 2024-05-09:
Great, thank you!

PR #6194 (2024-05-24): fix(sdl): handle if the window_id is not set correctly in SDL
fixes #6138

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py [... cut]
closed 2024-05-09 by kisvegabor