lvgl/lvgl #6139: lv_carousel_get_element_act is missing, how to get the carousel？

xinsuinizhuan, 2024-04-28:
### LVGL version

v9.1.1

### What happened?

in v8.3 have the lv_carousel_get_element_act function, but missing in v9， how to replace or get the carousel in v9?

kisvegabor, 2024-04-29:
There as no `lv_carousel` in v8 either. Which widget do you mean?

closed 2024-05-21 by lvgl-bot

ZhouKes, 2024-05-26:
I think your code is generated by NXP GUI Guider, the widgets 'carousel' is in NXP's LVGI Library, not in original LVGL; 
You can copy it to your projects. It's in \lvgl\src\extra\widgets\carousel

kisvegabor, 2024-05-26:
GuiGuider is not maintained by us but by NXP, so I can't comment on it. :slightly_frowning_face: 

(cc @nicusorcitu)

nicusorcitu, 2024-05-27:
Please address this question to NXP GG community:
https://community.nxp.com/t5/GUI-Guider/bd-p/GUI-Guider

nicusorcitu, 2024-05-30:
Please ask NXP Gui Guider team about the widgets, the best way to contact them is via NXP community:
https://community.nxp.com/t5/GUI-Guider/bd-p/GUI-Guider

nicusorcitu, 2024-05-30:
@cristian-stoica

xinsuinizhuan, 2024-05-30:
> Please ask NXP Gui Guider team about the widgets, the best way to contact them is via NXP community: https://community.nxp.com/t5/GUI-Guider/bd-p/GUI-Guider

I think it is about how to add the custome widget problem? need to pay attention to what