lvgl/lvgl #6140: how to get `connector_id` in lv_linux_drm_set_file on first time setup?

Coloryr, 2024-04-28:
### LVGL version

v9.1.0

### What happened?

`lv_linux_drm_set_file` is need `connector_id`, but I can't get it anywhere, it's bug?
I try 0, itn's work.

### How to reproduce?

```
    lv_init();

    lv_display_t * display = lv_linux_drm_create();

    lv_linux_drm_set_file(display, "/dev/dri/card0", 0);
```

Coloryr, 2024-04-28:
[image]

Coloryr, 2024-04-28:
If there is no first call to `lv_linux_drm_set_file`, then fb is -1, and the connector_id cannot be obtained. Only after calling `lv_linux_drm_set_file` can fb have a value and the `connector_id` can be obtained

kisvegabor, 2024-04-29:
@MiSimon could you take a look at this issue?

Coloryr, 2024-04-30:
I found that my device can pass 113, and this value will not change.

Coloryr, 2024-04-30:
But I encountered a more strange problem.

https://github.com/lvgl/lvgl/assets/26276037/180e430d-9fba-44a9-81d8-a954afc683ce

kisvegabor, 2024-05-02:
> But I encountered a more strange problem.

It's interesting because this issue usually happens in `LV_DISPLAY_RENDER_MODE_PARTIAL`when the rendered image is copied incorrectly. However the DRM driver uses `LV_DISPLAY_RENDER_MODE_DIRECT`. 

Maybe a color format mismatch? In `lv_linux_drm_set_file` please call `lv_display_set_color_fromat(disp, LV_COLOR_FORMAT_...)`  explicitly to be sure that the LVGL display's color format is correct.

Coloryr, 2024-05-07:
Only `LV_COLOR_FORMAT_NATIVE` display and refresh, other fromat can only display the first frame, or not display it,

MiSimon, 2024-05-07:
> The DRM driver was contributed @MiSimon. Michael, could you take a look at this issue?

@kisvegabor I was just the last one who touched it as I prepared the UEFI CI makefile. I think the code comes from the lv_drivers repo.

kisvegabor, 2024-05-07:
Sorry @MiSimon, I think @niklasf should be the one who can help us out. :crossed_fingers:

niklasf, 2024-05-08:
Hi. In most cases usage will look like:

```c
lv_display_t * display = lv_linux_drm_create();
lv_linux_drm_set_file(display, "/dev/dri/card0", -1 /* use first connector */);
```

If not, [`drm_info`](https://gitlab.freedesktop.org/emersion/drm_info) is a nice tool to enumerate connectors and their modes and properties. For example, the output might look like:

```
Node: /dev/dri/card0                <-- path
├───Driver: vc4 (Broadcom VC4 graphics) version 0.0.0 (20140616)
│   ├───DRM_CLIENT_CAP_STEREO_3D supported
│   ├───DRM_CLIENT_CAP_UNIVERSAL_PLANES supported
│   ├───DRM_CLIENT_CAP_ATOMIC supported
[...]
│   ├───DRM_CAP_ADDFB2_MODIFIERS = 1
│   ├───DRM_CAP_PAGE_FLIP_TARGET = 0
│   ├───DRM_CAP_CRTC_IN_VBLANK_EVENT = 1
│   ├───DRM_CAP_SYNCOBJ = 1
│   └───DRM_CAP_SYNCOBJ_TIMELINE = 0
├───Device: platform brcm,bcm2835-vc4
├───Framebuffer size
│   ├───Width: [0, 2048]
│   └───Height: [0, 2048]
├───Connectors
│   ├───Connector 0
│   │   ├───Object ID: 32          <-- the connector id i could have used instead of -1
│   │   ├───Type: HDMI-A
│   │   ├───Status: connected
│   │   ├───Physical size: 530x300 mm
│   │   ├───Subpixel: unknown
│   │   ├───Encoders: {0}
│   │   ├───Modes
│   │   │   ├───1920x1080@60.00 preferred driver phsync pvsync
│   │   │   ├───1920x1080@60.00 driver phsync pvsync 16:9
[...]
│   │   │   └───720x400@70.08 driver nhsync pvsync
│   │   └───Properties
│   │       ├───"EDID" (immutable): blob = 294
│   │       ├───"DPMS": enum {On, Standby, Suspend, Off} = On
│   │       ├───"link-status": enum {Good, Bad} = Good
│   │       ├───"non-desktop" (immutable): range [0, 1] = 0
│   │       ├───"TILE" (immutable): blob = 0
│   │       ├───"CRTC_ID" (atomic): object CRTC = 90
│   │       ├───"Colorspace": enum {Default, SMPTE_170M_YCC, BT709_YCC, XVYCC_601, XVYCC_709, SYCC_601, opYCC_601, opRGB, BT2020_CYCC, BT2020_RGB, BT2020_YCC, DCI-P3_RGB_D65, DCI-P3_RGB_Theater} = Default
│   │       ├───"left margin": range [0, 100] = 0
│   │       ├───"right margin": range [0, 100] = 0
│   │       ├───"top margin": range [0, 100] = 0
│   │       ├───"bottom margin": range [0, 100] = 0
│   │       ├───"max bpc": range [8, 12] = 8
│   │       ├───"Broadcast RGB": enum {Automatic, Full, Limited 16:235} = Automatic
│   │       └───"Output format": enum {Automatic, RGB, YCbCr 4:2:2, YCbCr 4:4:4} = Automatic
│   └───Connector 1
│       ├───Object ID: 57
[...]
```

kisvegabor, 2024-05-08:
Thank you for the info! I think it would be important to document for the future DRM users. Would you mind adding a brief guide about it [here](https://docs.lvgl.io/master/integration/driver/display/index.html)?

Coloryr, 2024-05-09:
Thanks, -1 is work, but still like this.

> But I encountered a more strange problem.
> 
>  VID_20240430_083854.1.mp4

closed 2024-05-30 by lvgl-bot