lvgl/lvgl #6142: rlottie animations not running as fast as they should

stwirth, 2024-04-28:
### LVGL version

8.3

### What happened?

When an rlottie animation has a higher framerate than the lvgl update rate (frequency of calls to `lv_timer_handler`) then it runs slower as it should.

stwirth, 2024-04-28:
The next frame callback is initialized using the frame rate specified in the lottie:
https://github.com/lvgl/lvgl/blob/36a805251034e55b4b5c087cab7e461ff7250727/src/extra/libs/rlottie/lv_rlottie.c#L148

and then in the callback, the frame number is simply incremented by 1:
https://github.com/lvgl/lvgl/blob/36a805251034e55b4b5c087cab7e461ff7250727/src/extra/libs/rlottie/lv_rlottie.c#L253-L254

I think it should instead calculate which frame should be rendered based on the time elapsed.

kisvegabor, 2024-04-29:
Hi,

Can you try it this PR? It uses a more recent Lottie implementation and it's also easier to integrate into LVGL.

stwirth, 2024-05-01:
Nice development @kisvegabor! Unfortunately I'm still on v8.3 so testing your PR will involve quite a bit of refactoring for me. I'll report back when I get it done.

closed 2024-05-30 by lvgl-bot