lvgl/lvgl #6145: Arabic and Hebrew percent sign display issues under RTL layout

qinshijing, 2024-04-29:
### LVGL version

v9.1.1

### What happened?

Both Arabic and Hebrew have RTL layouts. The percent sign in Hebrew is on the right side of the number, and the percent sign in Arabic is on the left side of the number. The RTL layout will force the percent sign to the left side of the number. as follows:
[image]
[image]

```C
const char* Arab_string =
"زيادة 56%";

const char* Hebrew_string = 
"עלייה של 56%.";

void lv_freetype_demo(void)
{
    /*Create a font*/
  lv_font_t* font = lv_freetype_font_create("./NotoSansArabic-Bold.ttf",
                                            LV_FREETYPE_FONT_RENDER_MODE_BITMAP,
                                            24, LV_FREETYPE_FONT_STYLE_NORMAL);

  lv_font_t* font2 = lv_freetype_font_create("./NotoSansHebrew-Bold.ttf",
                                            LV_FREETYPE_FONT_RENDER_MODE_BITMAP,
                                            24, LV_FREETYPE_FONT_STYLE_NORMAL);

  if (!font || !font2) {
    LV_LOG_ERROR("freetype font create failed.");
    return;
    }

    font->fallback = font2;

    lv_obj_set_style_base_dir(lv_screen_active(), LV_BASE_DIR_RTL, 0);

    /*Create style with the new font*/
    static lv_style_t style;
    lv_style_init(&style);
    lv_style_set_text_font(&style, font);
    lv_style_set_text_align(&style, LV_TEXT_ALIGN_CENTER);

    /*Create a label with the new style*/
    lv_obj_t * label = lv_label_create(lv_screen_active());
    lv_obj_add_style(label, &style, 0);
    lv_label_set_text(label, Arab_string);
    lv_label_set_long_mode(label, LV_LABEL_LONG_WRAP);
    lv_obj_set_width(label, 400);

    label = lv_label_create(lv_screen_active());
    lv_obj_add_style(label, &style, 0);
    lv_label_set_text(label, Hebrew_string);
    lv_label_set_long_mode(label, LV_LABEL_LONG_WRAP);
    lv_obj_set_width(label, 400);
    lv_obj_set_y(label, 100);
}
```

### How to reproduce?

lvgl adds a language setting and recognizes the percent sign as LV_BASE_DIR_RTL in Hebrew. Is it possible?

qinshijing, 2024-04-30:
> I wonder what's the logic behind it. The standard doesn't make a difference between languages. There are RTL, LTR, weak, strong and neutral characters. But I haven't seen something like % behaves differently language by language.
> 
> I've checked other translator tool and the result the same as in Google Translate. So there needs to be something that I/we don't know.
> 
> Arabic/Hebrew native speakers, please join. 🙂

I don't know why there is this difference, maybe it is a special language habit.

kisvegabor, 2024-05-15:
> I don't know why there is this difference, maybe it is a special language habit

It seems you are right. I've just approved the PR.

algovoid, 2024-05-22:
For Arabic, percentage sign is fixed on the left side of the digit: %10  and for Hebrew, percentage sign is on the right side of digit: 10%.

kisvegabor, 2024-05-26:
Thank you for confirming it. With https://github.com/lvgl/lvgl/pull/6146 we can set a custom list of neutral characters where `%` might or might not be included. 

As PR is merged, I close this issue.

closed 2024-05-26 by kisvegabor