lvgl/lvgl #6149: lv_obj_set_style_border_side() doesn't work!

aliparto, 2024-04-29:
### LVGL version

v9.1

### What happened?

The function "lv_obj_set_style_border_side" is utilized to adjust the border side, as illustrated in the [example](https://github.com/lvgl/lvgl/blob/release/v9.1/examples/widgets/tabview/lv_example_tabview_2.c). However, it encounters issues and fails to be applied to the buttons.

kisvegabor, 2024-04-30:
Please send a code snippet to reproduce the issue. Note that the slight gray line here is a shadow and not a border:
[image]

aliparto, 2024-04-30:
> Please send a code snippet to reproduce the issue. Note that the slight gray line here is a shadow and not a border: [...]

I tried to change the border side of the selected button of the **tab view** to the **right** but it didn't changed and stay at the bottom side (default theme value).
In v8.3 works fine.

kisvegabor, 2024-05-06:
Oh, I see now. This border is on the the button, but the container of the buttons. [This style](https://github.com/lvgl/lvgl/blob/master/src/themes/default/lv_theme_default.c#L587-L594) is applied on the tab view buttons.

aliparto, 2024-05-07:
Thanks, I got it.
Is it possible to customize the style of tab view buttons individually?
I mean if I want to have different tab views on various pages, how can I implement different styles for their respective buttons?

kisvegabor, 2024-05-07:
Yes, you can do this to get the buttons:
```c
    lv_obj_t * tab_bar = lv_tabview_get_tab_bar(tv);
    lv_obj_t * btn1 = lv_obj_get_child(tab_bar, 0);
    lv_obj_t * btn2 = lv_obj_get_child(tab_bar, 1);
```

kisvegabor, 2024-05-27:
This is a working example:
```c
    /*Create a Tab view object*/
    lv_obj_t * tabview;
    tabview = lv_tabview_create(lv_screen_active());

    /*Add 3 tabs (the tabs are page (lv_page) and can be scrolled*/
    lv_obj_t * tab1 = lv_tabview_add_tab(tabview, "Tab 1");
    lv_obj_t * tab2 = lv_tabview_add_tab(tabview, "Tab 2");
    lv_obj_t * tab3 = lv_tabview_add_tab(tabview, "Tab 3");

    lv_obj_t * tab_bar = lv_tabview_get_tab_bar(tabview);

    lv_obj_t * btn1 = lv_obj_get_child(tab_bar, 0);
    lv_obj_set_style_border_side(btn1, LV_BORDER_SIDE_RIGHT, LV_STATE_CHECKED);

    lv_obj_t * btn2 = lv_obj_get_child(tab_bar, 1);
    lv_obj_set_style_border_side(btn2, LV_BORDER_SIDE_RIGHT, LV_STATE_CHECKED);
    
    lv_obj_t * btn3 = lv_obj_get_child(tab_bar, 2);
    lv_obj_set_style_border_side(btn3, LV_BORDER_SIDE_RIGHT, LV_STATE_CHECKED);
```

closed 2024-06-20 by kisvegabor