lvgl/lvgl #6157: LVGL and MPLAB X project on Windows - GCC and CreateProcess issue

wavejaco, 2024-04-30:
### Introduce the problem

I am creating a basic MPLAB X project in Windows to include the use of LVGL. I have followed all the steps to set up the project as explained on the website.

When building the project, the build fails with the following error:
"make (e=87): The parameter is incorrect."

From reading up on the issue, it seems like this issue is triggered due to the Windows limitation of 32k on the CreateProcess function that is used during the project build. This also doesn't seem to be an issue limited to MPLAB X , as there seem to be various cases of this same issue in platforms such as Eclipse, NXP, etc.

Some online answers suggested that the project could be split up in libraries and built in stages, but I have no idea where to start with this approach. Others suggest that the project be built in Linux where this limit is not a problem. 

There must be a straight-forward solution to this issue, as I believe there are a lot of Windows users that use LVGL and have no issues with this.

I am attaching a text file with the complete build console output, as it is very long and I don't want to clutter this question.

Kindly advise on how to move forward with this.
[MPLAB X build issue console output.txt](https://github.com/lvgl/lvgl/files/15160291/MPLAB.X.build.issue.console.output.txt)

kisvegabor, 2024-05-02:
Hi,

Line 474 is really very long in the log. 

I haven't found any good workarounds either, but what you can do on LVGL's side is removing the `examples`, `demos` and `tests` folders  from the build. It will lower the number of files to build greatly.

wavejaco, 2024-05-22:
Okay, so some feedback on this issue. 

From an example project that I have been elsewhere, it seems that the main cause of the problem that I have experienced was that, as [outlined on the LVGL website](https://docs.lvgl.io/master/get-started/quick-overview.html), I added the entire "lvgl" folder to my project, where I was seemingly only supposed to add the "src" folder along with the additional header files. So it seems that MPLAB attempted to build all of the examples at once, which caused the issue as described in my initial post. This is in line with what @kisvegabor said above.

I can successfully build the project using only the necessary files for my project.

Perhaps this can be explained slightly more clearly on the website? It is somewhat misleading, as it was in my case.

kisvegabor, 2024-05-22:
Great that it's working now! I've open #6250 to describe which folders are needed in more detail.

PR #6250 (2024-05-27): docs(quick-overview): clarify which folders are required
fixes #6157

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and [... cut]
closed 2024-05-27 by FASTSHIFT