lvgl/lvgl #6166: `lv_obj_set_property()` with selector

hyansuper, 2024-05-02:
### Introduce the problem

can I set property for different state/part with code described in https://docs.lvgl.io/master/others/obj_property.html#a-step-further.
for example, I want to specify PART_INDECATOR color property in the array, how to do it?

hyansuper, 2024-05-02:
Oh, I think I figure it out:
```c
lv_property_t props[] = {
    { .id = LV_STYLE_BG_COLOR | LV_PART_INDICATOR, .color = (lv_color_t){.red = 0x11, .green = 0x22, .blue = 0x33}, },
}

LV_OBJ_SET_PROPERTY_ARRAY(obj, props);
```
This is really neat!

hyansuper, 2024-05-03:
There's a small error:
[LV_PROPERTY_ID_BUILTIN_LAST](https://github.com/lvgl/lvgl/blob/67fad672e3711013c6d6fa45ac3ce0ab6867f7f4/src/core/lv_obj_property.h#L61) should be larger than [LV_PROPERTY_IMAGE_END](https://github.com/lvgl/lvgl/blob/67fad672e3711013c6d6fa45ac3ce0ab6867f7f4/src/widgets/image/lv_image.h#L98)

kisvegabor, 2024-05-03:
Hi @hyansuper,

Thanks you for opening this issue.

> LV_STYLE_BG_COLOR | LV_PART_INDICATOR

I can't see in the code where/why it should work. See [here](https://github.com/lvgl/lvgl/blob/master/src/core/lv_obj_property.c#L61). 

@XuNeo Is the `set_property` feature still interesting for you? If so we can plan to finish it and make it more compete.

hyansuper, 2024-05-03:
> LV_STYLE_BG_COLOR | LV_PART_INDICATOR

my mistake, that won't work😂

XuNeo, 2024-05-04:
I have looked into the issue, and found out no easy solution. Let's discuss.
There are two main solutions.
1. add `selector` to  all property APIs. There could be requirement to specify `selector` in a const property array. Thus it may also required to add `selector` field in `lv_property_t`.
Note that `selector` is only used when id is style ID.
```c
typedef struct {
    lv_prop_id_t id;                /**< property ID, includes style ID(lv_style_prop_t) and widgets property ID.*/
    lv_style_selector_t selector;   /**< style selector, only used for when id is style ID*/
    union {
        int32_t num;                /**< Number integer number (opacity, enums, booleans or "normal" numbers)*/
        const void * ptr;           /**< Constant pointers  (font, cone text, etc)*/
        lv_color_t color;           /**< Colors*/
        lv_value_precise_t precise; /**< float or int for precise value*/
        lv_style_value_t _style;    /**< A place holder for style value which is same as property value.*/
    };
} lv_property_t;

lv_result_t lv_obj_set_property(lv_obj_t * obj, const lv_property_t * value, lv_style_selector_t selector);
lv_property_t lv_obj_get_property(lv_obj_t * obj, lv_prop_id_t id, lv_style_selector_t selector);

```
2. Remove style ID support from property API, `lv_obj_set_local_style_prop` can be used directly. Widget general properties and styles are two categories and they deserve two sets of API.
```c
lv_result_t lv_obj_set_property(lv_obj_t * obj, const lv_property_t * value);
void lv_obj_set_local_style_prop(lv_obj_t * obj, lv_style_prop_t prop, lv_style_value_t value,
                                 lv_style_selector_t selector);
```

I vote for 2, since mixing them looks really neat.

> Oh, I think I figure it out:
> 
> ```c
> lv_property_t props[] = {
>     { .id = LV_STYLE_BG_COLOR | LV_PART_INDICATOR, .color = (lv_color_t){.red = 0x11, .green = 0x22, .blue = 0x33}, },
> }
> 
> LV_OBJ_SET_PROPERTY_ARRAY(obj, props);
> ```
> 
> This is really neat!

hyansuper, 2024-05-04:
> lv_property_t props[] = {
    { .id = LV_STYLE_BG_COLOR | LV_PART_INDICATOR, .color = (lv_color_t){.red = 0x11, .green = 0x22, .blue = 0x33}, },
}

Do you mean to make this usable?
I like this solution too, you already mixed data type into id by bitwise operation, we can also mix in selectors

XuNeo, 2024-05-04:
> you already mixed data type into id by bitwise operation, we can also mix in selectors

The issue is that `selector` is 32bit long(state + part). So id alone is not enough.

> I am not sure, is `label_long_mode` considered a property or a style? to me it controls how label looks, just like label size.

It feels like style is a special property one widget can have. Widget can also have normal properties like label long mode etc.

> Do you mean to make this usable?

I mean that we should drop style prop ID support from API `lv_obj_set_property` and call `lv_obj_set_local_style_prop` (with some helpers) if you want to set a style value.

hyansuper, 2024-05-06:
Hi @XuNeo , I hope you keep the property API that unifies style and properties.

It's ok if selectors can't be used. I can add 'LV_PROPERTY_BAR_INDICATOR_COLOR' if I need it. just like you did in [lv_image.c](https://github.com/lvgl/lvgl/blob/37cf3bb4e41a1713d437de1090b7b5e37dbd4728/src/widgets/image/lv_image.c#L34).

With lv_obj_set_property(), I can make my device swap UI easily by parsing a portable bin file that contains a list of property id-value pairs.

XuNeo, 2024-05-21:
Not stale.

Hi @kisvegabor ,

What do you think of adding `selector ` filed to property struct and APIs?

XuNeo, 2024-05-21:
> Hi @XuNeo , I hope you keep the property API that unifies style and properties.

Since style is a special property, how about change the value of property for style, to include state and part selectors?
```c
typedef struct {
    lv_prop_id_t id;
    union {
        int32_t num;                /**< Number integer number (opacity, enums, booleans or "normal" numbers)*/
        const void * ptr;           /**< Constant pointers  (font, cone text, etc)*/
        lv_color_t color;           /**< Colors*/
        lv_value_precise_t precise; /**< float or int for precise value*/
        struct {
            uint32_t selector;
            lv_style_value_t _style;
        } style;
    };
} lv_property_t;
```

The struct size will 4bytes longer of cause.

hyansuper, 2024-05-22:
Sounds good to me😃

closed 2024-05-31 by XuNeo

hyansuper, 2024-07-20:
it seems the "31" in https://github.com/lvgl/lvgl/blob/70c562de16bb3a196c907ec99199e4824b18368c/src/core/lv_obj.h#L184 is a mistake @XuNeo

XuNeo, 2024-07-20:
Shoot. I'm making same mistake in the new PR. https://github.com/lvgl/lvgl/pull/6537

I'll correct them. Thanks!

hyansuper, 2024-07-21:
> There's a small error:
>
> [LV_PROPERTY_ID_BUILTIN_LAST](https://github.com/lvgl/lvgl/blob/67fad672e3711013c6d6fa45ac3ce0ab6867f7f4/src/core/lv_obj_property.h#L61) should be larger than [...]

Also this one @XuNeo

XuNeo, 2024-07-21:
It's already changed to a large value on master:

https://github.com/lvgl/lvgl/blob/70c562de16bb3a196c907ec99199e4824b18368c/src/core/lv_obj_property.h#L67

PR #6275 (2024-05-31): feat(property): add style selector support for property API
### Description of the feature or fix

Support style selector in property API.

1. Set property API remains same.
2. Introduce `lv_obj_get_style_property` for style property to support `selector`. 
3. Original `lv_obj_get_property` can also be used for style property but with `selector` set to 0.

Since the property value shares same union with `lv_style_value_t`, initialize the style property value is straight forward.

```c
static const lv_property_t obj_pos_x = {
     .id = LV_STYLE_X,
     .num = 123,
     .selector = LV_STATE_PRE [... cut]