lvgl/lvgl #6168: stm32u5g9j-dk1

OscarMonedero, 2024-05-02:
### Introduce the problem

Hi, i am trying to create a project with a stm32u5g9 board and lvgl v9, but i don´t know where i have to start or which are the drivers necessary for this. And i don´t find any documentation or examples on internet searchs, can someone help me?

kisvegabor, 2024-05-06:
Hi,

Can it be a good starting point? https://github.com/lvgl/lv_stm/tree/master/riverdi/lv_riverdi_stm32u5

OscarMonedero, 2024-05-07:
I'll take a look thanks

OscarMonedero, 2024-05-09:
I'm not sure this can help me because the display that i use is mipi dsi and i think the one in that project is a spi display.
I've been trying to change the files of hal_stm_lvgl folder from the stm32f769i_port but the configuration of this dsi seems to be so different. Do you have any documentation where i can search how to configure this type of display? 
Thanks!!

kisvegabor, 2024-05-13:
Unfortunately we don't have docs for that. 

Setting up MPIP DSI is not strictly related to LVGL. We are working on adding ready to use projects, but configuring the drivers/frame buffer is normally the users' task and more related to the chip vendors' docs or SDK. Sorry that I couldn't help more. :slightly_frowning_face:

closed 2024-06-10 by lvgl-bot