lvgl/lvgl #6172: Bar rotates automatically to horizontally if width>heigth

FloHomi, 2024-05-02:
### LVGL version

V9.1.1

### What happened?

Bar rotates automatically to horizontally if width>heigth
This should be a setting. User can not create a "FAT" bar. 
[image]

### How to reproduce?

    lv_obj_t* bar = lv_bar_create(water_cont);
    lv_obj_set_size(bar, 200, 100);

kisvegabor, 2024-05-06:
You are right! I think there should be a function like `lv_bar_set_orientation(bar, LV_BAR_ORIENTATION_HORIZONTAL/VERTICAL/AUTO)`

Do you have time to contribute with this feature?

liamHowatt, 2024-05-10:
@FloHomi you can test out @TridentTD's PR #6212 for your use case and provide feedback if you want, before it gets merged.

FloHomi, 2024-05-13:
Hi,
this looks great! Thank you :-) Taking some days off and all done already^^

closed 2024-06-10 by lvgl-bot

FloHomi, 2024-09-12:
I found an Issue.
the enums lv_bar_mode_t & lv_bar_orientation_t are both only 2 bit in size. Therfore the bar struct is implemented as such:

struct lv_bar_t {
    lv_obj_t obj;
    int32_t cur_value;          /**< Current value of the bar*/
    int32_t min_value;          /**< Minimum value of the bar*/
    int32_t max_value;          /**< Maximum value of the bar*/
    int32_t start_value;        /**< Start value of the bar*/
    lv_area_t indic_area;       /**< Save the indicator area. Might be used by derived types*/
    bool val_reversed;          /**< Whether value been reversed */
    lv_bar_anim_t cur_value_anim;
    lv_bar_anim_t start_value_anim;
    lv_bar_mode_t mode;     /**< Type of bar*/
    lv_bar_orientation_t orientation;  /**< Orientation of bar*/
};

but, in my visual studio simulator the enums are handled as int and this results in wrong behavior. when assigning a 2 to it (LV_BAR_ORIENTATION_VERTICAL = 2)
same for mode.

removing the ": 2" or 
uint8_t  mode instead of lv_bar_mode_t mode; solves the issue. I wish C would have type assignment to enums, as does c++....