lvgl/lvgl #6173: Compilation error using VG-Lite

oleg-amp, 2024-05-02:
### LVGL version

v9.1

### What happened?

When I set LV_USE_DRAW_VGLITE in lv_conf.h I get compilation error: `'vg_lite_int32_t' undeclared`
I can see that it is declared in vg_lite.h file located in vg_lite_tvg directory. 

Please elaborate how to compile the lvgl with vg_lite correctly. I did read all the related documentation. It does not explain well the required versions of lvgl <--> nxp SDK <--> vg_lite.

I use lvgl 9.1 with vg_lite took from nxp SDK 2.13.1 (vg_lite is version 2 there).

kisvegabor, 2024-05-03:
@anaGrad @nicusorcitu 
Could you take a look at this issue?.

josepho0918, 2024-05-07:
The same issue encountered.

nicusorcitu, 2024-05-08:
@cristian-stoica

nicusorcitu, 2024-05-08:
LVGL 9.1 works only with vglite 4.x

oleg-amp, 2024-05-08:
Hmm, so how am I supposed to use vg-lite on NXP RT1176 with lvgl 9.1?
Are there any instructions to use/port vg-lite 4.x for NXP?
I tried to find the answer in https://github.com/lvgl/lvgl/issues/4813 couldn't get to the bottom of the line where to take code/examples from.

nicusorcitu, 2024-05-08:
>> Hmm, so how am I supposed to use vg-lite on NXP RT1176 with lvgl 9.1?
>
You will need to wait for the SDK release 2.17.

oleg-amp, 2024-05-08:
What if I roll-back lvgl to 8.x, how do I use vg-lite 3.x? Which NXP SDK has the 3.x vg-lite?

nicusorcitu, 2024-05-08:
I would like to mention that SDK 2.13.1 that you are currently using already have vglite driver 3.0.15.

oleg-amp, 2024-05-08:
Ok, I checked the version in vg_lite.h, but it appears I had to check the ChangeLogKSDK.txt file. In vg_lite.h file version 2 is defined, while ChangeLogKSDK.txt shows current version is 3.0.15_rev7.
Ok, thanks for you help. Closing this issue. I will open new issue related to drawing lines with vg-lite to present why lvgl 9 may come handy compared to lvgl 8.

closed 2024-05-08 by oleg-amp