lvgl/lvgl #6183: Label of List can't be centered.

Kirbyrawr, 2024-05-05:
### LVGL version

9.1.0

### What happened?

[image]

As you can see in the image and the code shared below, the text won't center.

```
lv_obj_t * list = lv_list_create(content);
    lv_obj_add_style(list, &list_style, 0);
    lv_obj_set_size(list, LV_PCT(100), LV_PCT(100));

    const char *languages[] = {"Test", "Test", "Test", NULL};
    lv_obj_t* button;
    lv_obj_t* buttonLabel;

    for (const char **lang = languages; *lang != NULL; lang++) {
        button = lv_list_add_button(list, NULL, *lang);
        lv_obj_set_size(button, LV_PCT(100), 100);
        lv_obj_add_style(button, &button_style, 0);
    
        buttonLabel = lv_obj_get_child(button, 0);
        lv_obj_add_style(buttonLabel, &button_label_style, 0);
        lv_obj_set_size(buttonLabel, lv_pct(100), LV_SIZE_CONTENT);
        lv_obj_align(buttonLabel, LV_ALIGN_CENTER, 0, 0);
        lv_obj_set_style_bg_color(buttonLabel, lv_color_make(255, 0, 0), 0);
        lv_obj_set_style_bg_opa(buttonLabel, 255, 0);

        //lv_obj_add_event_cb(button, event_handler, LV_EVENT_CLICKED, NULL);
    }
```

The styles are just for borders and the text to be aligned horizontally.

```
 //Button
    lv_style_init(&button_style);
    lv_style_set_radius(&button_style, 10);
    lv_style_set_bg_color(&button_style, lv_color_make(255, 255, 255));
    lv_style_set_border_width(&button_style, 2);
    lv_style_set_border_side(&button_style, LV_BORDER_SIDE_FULL);
    lv_style_set_border_color(&button_style, lv_color_make(0, 0, 0));
    lv_style_set_pad_all(&button_style, 10);
    lv_style_set_align(&button_style, LV_ALIGN_CENTER);
    lv_style_set_text_align(&button_style, LV_TEXT_ALIGN_CENTER);

    //Button Label
    lv_style_init(&button_label_style);
    lv_style_set_text_align(&button_label_style, LV_TEXT_ALIGN_CENTER);
    lv_style_set_align(&button_label_style, LV_ALIGN_CENTER);
```

TridentTD, 2024-05-07:
```
    for (const char **lang = languages; *lang != NULL; lang++) {
        button = lv_list_add_button(list, NULL, *lang);
        lv_obj_set_size(button, LV_PCT(100), 100);
        lv_obj_set_flex_align(button, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);
    }
```

Kirbyrawr, 2024-05-07:
It works wonderful, thanks @TridentTD i thought it was a bug, closing the issue.

closed 2024-05-07 by Kirbyrawr