lvgl/lvgl #6192: LVGL Forum SPAMMER

PGNetHun, 2024-05-07:
### LVGL version

-

### What happened?

Hello!

A stupid spammer appeared in LVGL forum, please BAN that:
https://forum.lvgl.io/u/Kumari

PGNetHun, 2024-05-07:
Unfortunately scammers from India found LVGL forum (among other forums), and now they are bombing it with messages.
@kisvegabor , if you wish I can help deleting those topics/messages and banning those scammers, or if it's possible write a script that filters out those scam messages before being stored in forum database. (filter using regular expressions or simply checking for some keywords)

kisvegabor, 2024-05-07:
Thank you for offering your help! Is it possible to write a custom script to Discourse?

I also added some filters and disabled google based login.

kisvegabor, 2024-05-07:
They are still passing through...

Can you link some other forums where they are attacking?

kisvegabor, 2024-05-07:
I have installed the [Akismet](https://akismet.com/) plugin. Hopefully it will make it.

PGNetHun, 2024-05-12:
> I have installed the [Akismet](https://akismet.com/) plugin. Hopefully it will make it.

Since then there are no spam messages in forum.
Thx, I will close this ticket.

closed 2024-05-12 by PGNetHun

kisvegabor, 2024-05-13:
Yeah, it seems it really solved it.