lvgl/lvgl #6193: Object Rotation

fatihozen3, 2024-05-07:
### Introduce the problem

I use IMXRT1170 MCU and LVGL v9.1.0. I have a simple problem but I don't know the reason.  I want to see my image rotated with lv_img_set_angle function but I face with MemManage_Handler. What could be the reason ?

            img_test_1 = lv_img_create(screen_test);
            lv_obj_add_flag(img_test_1, LV_OBJ_FLAG_CLICKABLE);
            lv_img_set_src(img_test_1, "F:/sample.bin");
            lv_img_set_pivot(img_test_1, 50, 50);
            lv_img_set_angle(img_test_1, 100);
            lv_obj_set_pos(img_test_1, 100, 100);
            lv_obj_set_size(img_test_1, 64, 69);
            
            
[image]

liamHowatt, 2024-05-09:
Hi, what version are you using (v8, v9, etc.)?

When I try your to rotate a binary image it doesn't appear correctly, so that's definitely a concern.

However I do not get a memory error. I think an error is being encountered and the program proceeds with a `NULL` or otherwise invalid value. Could you please enable `LV_LOG_LEVEL_WARN` and `LV_USE_ASSERT_NULL`, `LV_USE_ASSERT_MALLOC` if not already enabled?

Also, does it work with the binary image when it's not rotated?

fatihozen3, 2024-05-09:
> Hi, what version are you using (v8, v9, etc.)?
>
> When I try your to rotate a binary image it doesn't appear correctly, so that's definitely a concern.
>
> However I do not get a memory error. I think an [...]

Hi, I use v9.1.0, I can see binary images when it is not rotated, but I have not tried unbinary pictures. All things are enabled as you suggested

fatihozen3, 2024-05-09:
I have added it as c array not binary and it is rotated with lv_img_set_angle function successfully. But when I try it with binary image, no image is shown on the screen. 

Second issue is angle value. For one picture when I write there 434, I see MemManageHandler fault, but when I write 435, I don't see this fault, screen can load but image doesn't shown in screen.

I have tried these 434,435 values for different image, it does not work. For both value I see MemManageHandler so it is related with picture(some of it's parameters maybe width height). But for sure I could not see rotated binary image on the screen.

fatihozen3, 2024-05-09:
I use ARGB8888 color format. There is written sth useful [here](https://docs.lvgl.io/9.1/widgets/image.html)
The transformations require the whole image to be available. Therefore indexed images (LV_COLOR_FORMAT_I1/2/4/8_...), alpha only images cannot be transformed. In other words transformations work only on normal (A)RGB or A8 images stored as C array, or if a custom [Image decoder](https://docs.lvgl.io/9.1/overview/image.html#overview-image-decoder) returns the whole image.

liamHowatt, 2024-05-10:
Ah, so it's not supported for now. Good find. I think some undefined behavior is reasonable when you try to transform it, then. I could not reproduce the `lv_img_set_angle(img_test_1, 434)` crash but it's not worth fixing until it's supposed to be supported.

The situation right now is a bit weird.

|  | bin decoder | png decoder |
|---|---|---|
| decodes whole image | 👎️  | ✅️ |
| transform works | 👎️  | ✅️ |
| transform works if whole image was cached | 👎️  | N/A |

The bin decoder decoding incrementally is supposed to be helpful, but it actually punishes users who want to transform the image. Even if the image cache is big enough to cache the whole image after the first decode, it still can't be transformed even though the whole image is now available.

I think something could be done about this.

For your requirement of needing to load an image dynamically, there are a few options! :slightly_smiling_face: The simplest is to just store the image on your storage device as PNG! I'd recommend enabling image caching (`LV_CACHE_DEF_SIZE` and `LV_IMAGE_HEADER_CACHE_DEF_CNT` in lv_conf.h) so it doesn't read the file every time. Another option is to store the raw image data from the C array in a binary file, and then `lv_fs_read` it and initialize the fields of an `lv_image_dsc_t` manually.

closed 2024-06-10 by lvgl-bot