lvgl/lvgl #6196: Incomplete implementation of lv_canvas_set_px() for indexed color format

vitotai, 2024-05-08:
### LVGL version

v9.1

### What happened?

https://github.com/lvgl/lvgl/blob/46887dbe5149616ba8cebeb45039880d462b3519/src/widgets/canvas/lv_canvas.c#L113

It only works for 1bit indexed format, not for others.

### How to reproduce?

        uint32_t px = ((dsc->header.w + 7) >> 3) * y + x;
        buf_u8[px]  = buf_u8[px] & ~(1 << (7 - bit));
        buf_u8[px]  = buf_u8[px] | ((c.full & 0x1) << (7 - bit));

liamHowatt, 2024-05-09:
Hi, thanks for the tip. I'm not seeing it work even for `I1`.

From left to right: `LV_COLOR_FORMAT_I1`, `LV_COLOR_FORMAT_I2`, `LV_COLOR_FORMAT_I4`, `LV_COLOR_FORMAT_I8`, `LV_COLOR_FORMAT_L8` (`L8` works).

[image]

kisvegabor, 2024-05-09:
It's [implemented](https://github.com/lvgl/lvgl/blob/master/src/widgets/canvas/lv_canvas.c#L113-L124) but really seems related to 1 bit only. 

I think it's not too difficult to fix it. The only trick is that the palette at the beginning needs to be skipped (`buf += 8;`)

@vitotai would you be interested in sending a PR?

vitotai, 2024-05-13:
@kisvegabor 

I am not good at operating those GitHub.com operations. It's a simple fix for the developers of LVGL given the fact that there is similar code in V8.

I just wanted to remind you.

kisvegabor, 2024-05-14:
I couldn't resist and opened #6226 to fix it. Could you test it?

closed 2024-05-21 by FASTSHIFT