lvgl/lvgl #6198: The display will not receive the input event like press and release if a new screen is loaded.

sliang951753, 2024-05-08:
### LVGL version

v9.0.0

### What happened?

When I use the following code to add callback to the display
lv_display_add_event_cb(disp, display_event_handler, LV_EVENT_PRESSED, nullptr);
The callback will not be invoked if I create a screen object and load it.

It looks if I create a screen object using lv_obj_create(NULL), this object will be the root object, and the mouse event will not deliver to the display object. 

The display driver that I'm using is SDL driver.

liamHowatt, 2024-05-09:
Hey, I can't seem to get the callback to fire even before creating more screens. Could you please share a code snippet that works for getting the event? I'm also using SDL.

kisvegabor, 2024-05-09:
`LV_EVENT_PRESSED` is not propagated to displays. It shall be used with a widget. E.g.
```c
lv_obj_add_event_cb(scr, event_handler, LV_EVENT_PRESSED, nullptr);
```

sliang951753, 2024-05-10:
> `LV_EVENT_PRESSED` is not propagated to displays. It shall be used with a widget. E.g.
> 
> ```c
> lv_obj_add_event_cb(scr, event_handler, LV_EVENT_PRESSED, nullptr);
> ```

However, if I want to handle the global touch event, I have to put my code to every created screen. 
Another use case is I can easily hook the touch event on display if it can handle the touch event, and then I can implement the disable / enable touch feature.  I can prevent the touch event being dispatched to the screens. 

Although I can directly get the touch event from the indev, but I think its better to make all application code to depends on LVGL library so that it can be cross-platform.

sliang951753, 2024-05-10:
> Hey, I can't seem to get the callback to fire even before creating more screens. Could you please share a code snippet that works for getting the event? I'm also using SDL.

Sorry for confuse you, actually it doesn't work. The display object cannot receive the touch event as kisvegabor metioned.

sliang951753, 2024-05-10:
I found an alternative way to receive the global touch event. That is lv_indev_add_event_cb(). So, I closed this issue.

closed 2024-05-10 by sliang951753