lvgl/lvgl #6200: Compilation error using PXP

oleg-amp, 2024-05-08:
### LVGL version

v8.x and v9.x

### What happened?

When I enable flag `LV_USE_GPU_NXP_PXP` I get following error:
```
src/draw/nxp/pxp/lv_draw_pxp_blend.c:57:37: note: in definition of macro 'PXP_PS_PIXEL_FORMAT'
```
I use NXP MIMXRT1176 MCU and it has neither `FSL_FEATURE_PXP_HAS_NO_EXTEND_PIXEL_FORMAT` nor `FSL_FEATURE_PXP_V3` flags defined. It also does not have the `kPXP_PsPixelFormatARGB8888` enum value defined either. So what I do to solve this is everytime I clone lvgl code, I rename `kPXP_PsPixelFormatARGB8888` to `kPXP_PsPixelFormatRGB888` and everything works.

**Note:** In lvgl 9.x I need to rename more than 1 instances in the same file.

### How to reproduce?

```c
#define LV_USE_GPU_NXP_PXP 1
```

liamHowatt, 2024-05-09:
Thanks for the info. @nicusorcitu FYI. Any thoughts about this?

kisvegabor, 2024-05-09:
Maybe related to [this](https://github.com/lvgl/lvgl/issues/6173#issuecomment-2099966174).

oleg-amp, 2024-05-10:
> Maybe related to [this](https://github.com/lvgl/lvgl/issues/6173#issuecomment-2099966174).

No. This happens whether vglite is enabled or not, on both lvgl versions 8.x and 9.x.

nicusorcitu, 2024-05-10:
@cristian-stoica please have a look.

nicusorcitu, 2024-05-10:
What SDK version do you have?

oleg-amp, 2024-05-10:
2.13.1

nicusorcitu, 2024-05-10:
Why do you keep staying with the older SDK? Please take the latest from NXP website for MIMXRT1176 board and give it a try.

cristian-stoica, 2024-05-10:
@oleg-amp I've build tested an example from SDK2.13.1 and it builds fine. In fact LV_USE_GPU_NXP_PXP is enabled by default in the SDK example. I din not make other changes in the code.
For debug purposes I will need an example project of yours if you can provide one. I'm using MCUExpresso but if you're using something else that's fine.

oleg-amp, 2024-05-12:
> Why do you keep staying with the older SDK? Please take the latest from NXP website for MIMXRT1176 board and give it a try.

We have freezed the sdk in out project, for now, to avoid upgrading issues.

> I've build tested an example from SDK2.13.1 and it builds fine. In fact LV_USE_GPU_NXP_PXP is enabled by default in the SDK example. I did not make other changes in the code.
For debug purposes I will need an example project of yours if you can provide one. I'm using MCUExpresso but if you're using something else that's fine.

I am using NXP MCUExpresso plugin for VSCode and nsp-sdk taken from official github repo. As you can see [here](https://github.com/nxp-mcuxpresso/mcux-sdk/blob/MCUX_2.13.1_aarch64_0.2/drivers/pxp/fsl_pxp.h#L338), there is no definition of `kPXP_PsPixelFormatARGB8888` at all in the PXP driver.
In addition, I clone lvgl directly from github, while in MCUExpresso IDE they take cloned and modified version of lvgl. See the `src/draw/nxp/pxp/lv_draw_pxp_blend.c` as it is in MCUExpresso:
```c
/**
 * @file lv_draw_pxp_blend.c
 *
 */

/**
 * MIT License
 *
 * Copyright 2020-2022 NXP
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights to
 * use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
 * the Software, and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice (including the next paragraph)
 * shall be included in all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED,
 * INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A
 * PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
 * HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF
 * CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE
 * OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 */

/*********************
 *      INCLUDES
 *********************/

#include "lv_draw_pxp_blend.h"

#if LV_USE_GPU_NXP_PXP

/*********************
 *      DEFINES
 *********************/

#if LV_COLOR_16_SWAP
    #error Color swap not implemented. Disable LV_COLOR_16_SWAP feature.
#endif

#if LV_COLOR_DEPTH==16
    #define PXP_OUT_PIXEL_FORMAT kPXP_OutputPixelFormatRGB565
    #define PXP_AS_PIXEL_FORMAT kPXP_AsPixelFormatRGB565
    #define PXP_PS_PIXEL_FORMAT kPXP_PsPixelFormatRGB565
#elif LV_COLOR_DEPTH==32
    #define PXP_OUT_PIXEL_FORMAT kPXP_OutputPixelFormatARGB8888
    #define PXP_AS_PIXEL_FORMAT kPXP_AsPixelFormatARGB8888
    #define PXP_PS_PIXEL_FORMAT kPXP_PsPixelFormatRGB888
#elif
    #error Only 16bit and 32bit color depth are supported. Set LV_COLOR_DEPTH to 16 or 32.
#endif

#if defined (__alpha__) || defined (__ia64__) || defined (__x86_64__) \
    || defined (_WIN64) || defined (__LP64__) || defined (__LLP64__)
    #define ALIGN_SIZE 8
#else
    #define ALIGN_SIZE 4
#endif

/**********************
 *      TYPEDEFS
 **********************/
.
.
.
```

It looks like NXP fixed this issue in the integrated version of LVGL in the SDK for MXUExpresso, however if working with VSCode and manually cloning both the SDK and LVGL from repos this issue persists.

cristian-stoica, 2024-05-13:
> It looks like NXP fixed this issue in the integrated version of LVGL in the SDK for MXUExpresso, however if working with VSCode and manually cloning both the SDK and LVGL from repos this issue persists.

The reported build issues may be related to the SDK used - not necessarily the version but the origin. Let me suggest that you build and use the SDK at https://mcuxpresso.nxp.com/en/welcome (for MCUExpresso or VScode) instead of the one at [github ](https://github.com/NXP-mcuxpresso/mcux-sdk).
Integration of LVGL in [github ](https://github.com/NXP-mcuxpresso/mcux-sdk) sdk is not done yet. Because of this, there are no LVGL examples and you will need to fix all sorts of problems yourself if you use it for LVGL.

Indeed, there is also the issue of what LVGL version is integrated with the SDK and that it is a stripped down and modified package: SDK LVGL may have some patches that are not yet unstreamed for that particular release; the stripping down is for packaging reasons.
If you want to use a different snapshot of LVGL you can drop-replace the lvgl folder in the SDK with a different one as long as you preserve the major version (8 or 9). If the SDK uses LVGL 8 and you put LVGL 9, the build will probably fail and you will need to resolve the integration yourself (it can be done but...) or switch to an SDK that uses LVGL 9.

oleg-amp, 2024-05-15:
Because I use VSCode with the nxp official pluggin and cmake. I clone the lvgl code and build it using cmake fetchContent. It is too much work to manually add all the source code to the project cmakelists.txt.
Of course the integrated lvgl instance must support the nxp-sdk it is integrated within, but it is expected that the official repo of lvgl will support the official repo of the nxp-sdk, no? Otherwise, what is the point of having draw/nxp folder in lvgl repo if it does not support nxp-sdk repo? In any case, from what I saw, it is not documented in any place that it doesn't..

nicusorcitu, 2024-05-16:
> Of course the integrated lvgl instance must support the nxp-sdk it is integrated within, but it is expected that the official repo of lvgl will support the official repo of the nxp-sdk, no? Otherwise, what is the point of having draw/nxp folder in lvgl repo if it does not support nxp-sdk repo?
>
nxp-sdk repo will support lvgl v9, but starting with next sdk 2.17 release.

Currently, the LVGL middleware support into [nxp-sdk](https://github.com/nxp-mcuxpresso/mcux-sdk) is not yet done. Because of that, the github nxp-sdk repo might not have the proper PXP drivers.

If you want to build SDK 2.13 with LVGL + PXP, you need to use the archive generated SDK from here: https://mcuxpresso.nxp.com/en/welcome with all dependency drivers.

oleg-amp, 2024-05-16:
Thanks a lot for detailed explanation! 
Do you maybe know when nxp plans to release 2.17 sdk?

oleg-amp, 2024-05-16:
Ok, thanks! I won't wait for it of course due to planned timeline. 
I will leave it as it is for now, or will try to use the generated sdk from nxp website as you suggested.

closed 2024-05-16 by oleg-amp