lvgl/lvgl #6201: Draw line using VG-Lite

oleg-amp, 2024-05-08:
### LVGL version

v8.3.1

### What happened?

I want to draw multiple lines using GPU. I use NXP MIMXRT1176 which has VG-Lite.
The final result shall be an arc, drawn using short thin lines, like ticks, with constant radius from the center. I took the inspiration from the lv_meter widget.

To draw lines, I use inner and outer circular masks and draw the lines started from the center of the circle and end up right about the outer mask. This method draws all those pixels which are in-between the two circular masks.

I draw these lines upon changes in encoder, like a volume control. So I need these lines to be drawm efficiently and fast enough. My problem is that using lvgl 8.3.1 I cannot do it at all using VG-Lite because of the following code in lv_draw_vglite.c:
```c
static void lv_draw_vglite_line(lv_draw_ctx_t * draw_ctx, const lv_draw_line_dsc_t * dsc, const lv_point_t * point1,
                                const lv_point_t * point2)
{
    ...

    bool done = false;
    bool has_mask = lv_draw_mask_is_any(&rel_clip_area);

    if(!has_mask) {
        done = (lv_gpu_nxp_vglite_draw_line(&rel_point1, &rel_point2, &rel_clip_area, dsc) == LV_RES_OK);
        if(!done)
            VG_LITE_LOG_TRACE("VG-Lite draw line failed. Fallback.");
    }

    if(!done)
        lv_draw_sw_line(draw_ctx, dsc, point1, point2);
}
```
Meaning, if I have any masks at all, LVGL uses software to draw the lines instead of GPU. **Software method is extremelly slow**.

In LVGL 9 I didn't see such condition, so, I tried to to use it. Unfortunatelly, LVGL 9.x does not support VG-Lite 3.x, which basically is supported currentnly by the latest NXP SDK. 

What are my options (except maybe waiting for the NXP SDK which is planned to support VG-Lite 4.x)?

liamHowatt, 2024-05-09:
Hi. Are you able to draw the line segments without a mask? i.e. start the lines from where the inner mask's edge would be and end the lines where the outer mask's edge would be? If it's too difficult to align the ends of the lines nicely, could you draw something else like a `LV_DRAW_TASK_TYPE_ARC` where the mask would have been, on top of the lines?

kisvegabor, 2024-05-09:
> What are my options (except maybe waiting for the NXP SDK which is planned to support VG-Lite 4.x)?

See [this issue](https://github.com/lvgl/lvgl/issues/6173#issuecomment-2099966174) for comments from NXP on this topic.

> I draw these lines upon changes in encoder, like a volume control. So I need these lines to be drawm efficiently and fast enough.

What about having an image with the ticks and setting it as the background of an arc. The arc will mask out parts of the image as you adjust it's angles/value. It should be much faster than drawing the ticks one by one.

oleg-amp, 2024-05-10:
> What about having an image with the ticks and setting it as the background of an arc. The arc will mask out parts of the image as you adjust it's angles/value. It should be much faster than drawing the ticks one by one.

The problem is that the design should have one color for ON ticks and another for OFF tick. I solved this by drawing a line with a different color on a canvas on top of already drawn lines.

kisvegabor, 2024-05-13:
Can you send an image about how the result should look like?

oleg-amp, 2024-05-15:
[image]

kisvegabor, 2024-05-20:
You can try this:
```c
lv_obj_set_style_arc_image_src(arc, &img, LV_PART_MAIN);
lv_obj_set_style_arc_opa(arc, LV_OPA_50, LV_PART_MAIN);
lv_obj_set_style_arc_image_src(arc, &img, LV_PART_INDICATOR);
```

closed 2024-06-16 by lvgl-bot