lvgl/lvgl #6203: Scrolling does not respect padding of parent.

di-jtrumpower, 2024-05-08:
### LVGL version

v9.1.0-2-gcac17b580

### What happened?

Based on the documentation for scrolling:

"if an object is outside its parent content area (the size without padding), the parent becomes scrollable and scrollbar(s) will appear."

I am trying to create a basic container that has a column of elements to scroll through, the container has a border. The items are always being drawn over the container borders no matter the padding.

This seems like a pretty normal use case of scrolling, so I am not sure if I am overlooking something really simple here.

### How to reproduce?

```
lv_obj_t* scrollbox = lv_obj_create(screen);
lv_obj_set_style_pad_all(scrollbox, 15, LV_PART_MAIN);
lv_obj_set_style_bg_color(scrollbox,lv_color_hex(0xF0F0F0),LV_PART_MAIN);
lv_obj_center(scrollbox);
lv_obj_set_size(scrollbox,200,200);

lv_obj_set_flex_flow(scrollbox,LV_FLEX_FLOW_COLUMN);
lv_obj_set_flex_align(scrollbox, LV_FLEX_ALIGN_START, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);

lv_obj_set_style_radius(scrollbox, 10, LV_PART_MAIN);
lv_obj_set_style_border_color(scrollbox, lv_color_hex(0xED1C24), LV_PART_MAIN);
lv_obj_set_style_border_width(scrollbox, 4, LV_PART_MAIN);

lv_obj_set_scroll_snap_y(scrollbox, LV_SCROLL_SNAP_CENTER);    
lv_obj_set_style_width(scrollbox,8,LV_PART_SCROLLBAR);
lv_obj_set_style_radius(scrollbox, 4,LV_PART_SCROLLBAR);

lv_obj_t* button[10];

for(int i = 0; i < 10; i++)
{
  button[i] = lv_obj_create(scrollbox);
  lv_obj_t* label = lv_label_create(button[i]);
  lv_label_set_text_fmt(label,"%d",i);
  lv_obj_center(label);
}
```

liamHowatt, 2024-05-09:
Hey. Is this what you're seeing? Or something different? The contents don't appear to be drawn over the borders to me. Thanks for the code snippet!

[image]

di-jtrumpower, 2024-05-09:
Thanks for taking time to look at this,

I am not using the premade styles but, yes I am seeing it differently. This is on an stm32L4R9 but I don't think that would matter.

[image]

di-jtrumpower, 2024-05-09:
Just changed it from LV_USE_THEME_SIMPLE to LV_THEME_DEFAULT, and it no longer is rendering over the border. There must be some padding style I am missing.

liamHowatt, 2024-05-10:
These seem to work!
```c
lv_obj_set_style_border_post(scrollbox, true, LV_PART_MAIN); /* stops the borders being drawn over */
lv_obj_set_style_pad_all(scrollbox, 8, LV_PART_SCROLLBAR); /* moves the scrollbar inward, off the border */
```

di-jtrumpower, 2024-05-10:
Thanks, I knew it was something simple!

closed 2024-05-10 by di-jtrumpower