lvgl/lvgl #6209: How to display Thai on the interface

jeremyvirtue, 2024-05-09:
### LVGL version

9.0.0

### What happened?

when Thai is displayed with consonants superimposed on vowels and tones, the tone cannot be fully displayed and will be mixed with vowels,They show left and right. How can I display it correctly?

trivij, 2024-05-10:
use BlynkGO is the best!

liamHowatt, 2024-05-13:
Thanks for the tip @trivij!

@jeremyvirtue, which font were you seeing the issues with?

cc @BlynkGo (they participate here sometimes)

Regarding BlynkGO, I tried this:
https://github.com/BlynkGO/BlynkGO_font/blob/master/Eng-Thai/ThaiSansNeue/ThaiSansNeue_Bold_25.c
and it looks correct at first sight but I am not familiar with the language.

[image]

I simply downloaded the C file, added it to my build files, and made this change

```diff
diff --git a/../../Downloads/ThaiSansNeue_Bold_25.c b/../lvgl_workspace/ThaiSansNeue_Bold_25.c
index 49716e5b3..f3034a328 100644
--- a/../../Downloads/ThaiSansNeue_Bold_25.c
+++ b/../lvgl_workspace/ThaiSansNeue_Bold_25.c
@@ -4,7 +4,11 @@
  * Opts: --bpp 4 --size 25 --no-compress --font ThaiSansNeue-Bold_modified.ttf --range 32-126,161-255,3584-3675,8192-8805,63232-63632 --format lvgl -o ThaiSansNeue_Bold_25.c
  ******************************************************************************/
 
-#include "blynkgo_lib.h"
+#ifdef LV_LVGL_H_INCLUDE_SIMPLE
+    #include "lvgl.h"
+#else
+    #include "lvgl/lvgl.h"
+#endif
 
 #ifndef THAISANSNEUE_BOLD_25
 #define THAISANSNEUE_BOLD_25 1
```

And then used it like so:

```c
static void thai_font_blynkgo()
{
    LV_FONT_DECLARE(ThaiSansNeue_Bold_25);
    lv_obj_t * label = lv_label_create(lv_screen_active());
    lv_obj_set_style_text_font(label, &ThaiSansNeue_Bold_25, 0);
    lv_label_set_text(label, "วิธีการใช้งาน");
}
```

I also tried arluck_25.c but had a less satisfying result.

jeremyvirtue, 2024-05-14:
> ```diff
> blynkgo_lib
> ```
I try to understand,But do you have to use BlynkGO OS for this solution?Can we try to isolate it?

liamHowatt, 2024-05-14:
I didn't use BlynkGO OS. Please read my comment again :slightly_smiling_face: 

> I simply downloaded the C file, added it to my build files, and made this change

closed 2024-06-09 by lvgl-bot

LEKPCSTEAM, 2026-09-01:
ลองดูนะครับสำหรับแก้ไขสระลอย **ภาษาไทย** โดยเฉพาะใน .ttf  เมื่อแปลงเสร็จจะได้ .ttf ที่แก้ไขแล้ว รายละเอียดอยู่ใน repo ครับ 

Feel free to check this out—it's designed specifically for fixing **Thai** floating vowels in .ttf files. Once converted, you’ll get the patched .ttf file. All the details are in the repo. 

Before:

[image]

After:

[image]

Repo:
[https://github.com/LEKPCSTEAM/LVGL-Thai-Font-Tool.git](
https://github.com/LEKPCSTEAM/LVGL-Thai-Font-Tool.git)