lvgl/lvgl #6211: ESP32 Rotate doesn't work

hoanglam9925, 2024-05-09:
### LVGL version

v9.1.0

### What happened?

The lv_demo_benchmark demos work with esp32 and ST7789 when I use the TFT_eSPI library, however if I use tft.setRotation(4) to rotate the screen 90 degrees, the display breaks. When lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90/180) is used instead of tft.setRotation, the functionality is not achieved.
[image]

### How to reproduce?

#include <lvgl.h>
#include <TFT_eSPI.h>

#include <examples/lv_examples.h>
#include <demos/lv_demos.h>

/*Set to your screen resolution*/
#define TFT_HOR_RES     240
#define TFT_VER_RES     240

/*LVGL draw into this buffer, 1/10 screen size usually works well. The size is in bytes*/
#define DRAW_BUF_SIZE   (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))
void * draw_buf;
unsigned long lastTickMillis = 0;
static lv_display_t * disp;

TFT_eSPI tft = TFT_eSPI();

void setup()
{

    String string_LVGL_Arduino = "Hello Arduino!";
    string_LVGL_Arduino += String('V') + lv_version_major() + "." + lv_version_minor() + "." + lv_version_patch();

    Serial.begin(115200);
    Serial.println(string_LVGL_Arduino);

    lv_init();

    draw_buf = heap_caps_malloc(DRAW_BUF_SIZE, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL) ;

    /*TFT_eSPI can be enabled lv_conf.h to initialize the display in a simple way*/
    disp = lv_tft_espi_create(TFT_HOR_RES, TFT_VER_RES, draw_buf, DRAW_BUF_SIZE);

    //tft.setRotation(4);
    // lv_disp_set_rotation(disp, LV_DISPLAY_ROTATION_180);

    //lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_0);
    lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90); 
    // lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_180);
    //lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_270);

    //Drawing code
     lv_demo_benchmark();
//    lv_demo_stress();
    //lv_demo_scroll();
    
    Serial.println("LVGL Setup done");
}

void loop()
{
    unsigned int tickPeriod = millis() - lastTickMillis;
    lv_tick_inc(tickPeriod);
    lastTickMillis = millis();
    lv_task_handler(); /* let the GUI do its work */
    
    //delay(5); /* let this time pass */
}

liamHowatt, 2024-05-10:
Hmm that should be working as of #6001 which updated `lv_tft_espi` to correctly handle `lv_display_set_rotation`.

The problem could be that two instances of `TFT_eSPI` being created. You declare one in your program and `lv_tft_espi` also creates its own internally. Could be unrelated to the problem, though.

Use this Arduino example as a reference. It uses `lv_tft_espi`: https://github.com/lvgl/lvgl/blob/master/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino

hoanglam9925, 2024-05-11:
ok, I was able to rotate it, but I discovered that release 9.1 in the lib manager is the old version, I have to clone it myself to be correct. According to release v9.1 it still doesn't work. Tks

closed 2024-05-11 by hoanglam9925

liamHowatt, 2024-05-17:
Oops, you're right, that change got merged after v9.1.0.

vishvanathdoijad, 2024-08-08:
Is "sw_rotate" need to set in LVGL 9.1.1 which is in "lvgl/scr/display/lv_display_private.h"
If yes, Please let me know the steps how...?

liamHowatt, 2024-08-13:
This is the line that does it: https://github.com/lvgl/lvgl/blob/90cfb6c61ba612e89afbbad79cdd34f08e724d91/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino#L99

using `lv_display_set_rotation`.

`TFT_ROTATION` is defined here: https://github.com/lvgl/lvgl/blob/90cfb6c61ba612e89afbbad79cdd34f08e724d91/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino#L21

It can be `LV_DISPLAY_ROTATION_0`, `LV_DISPLAY_ROTATION_90`, `LV_DISPLAY_ROTATION_180`, or `LV_DISPLAY_ROTATION_270`.

vishvanathdoijad, 2024-08-13:
> This is the line that does it:
>
> https://github.com/lvgl/lvgl/blob/90cfb6c61ba612e89afbbad79cdd34f08e724d91/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino#L99
>
> using `lv_display_set_rotation`. [...]

Thanx for your reply.

Somehow, re-configuring the hardware rotation by using "lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_0/90/180/270)" is not work for me.
So my approach is used "lv_draw_sw_rotate()" in flush_cb() function, Is it correct approach? Please review my below flush_cb()

void lvgl_flush_cb(lv_display_t *disp_drv, const lv_area_t *area, lv_color_t * colour_p)
{
        int32_t x,y;

        // Create a buffer to hold the rotated image data
        lv_color_t rotated_buf[(rotated_w * rotated_h * sizeof(lv_color_t))];
        static lv_coord_t rotated_w = 0, rotated_h = 0;

        rotated_w = area->x2 - area->x1 + 1;
        rotated_h = area->y2 - area->y1 + 1;

        // "rotated_buf" contain rotated data
        lv_draw_sw_rotate(colour_p, rotated_buf, rotated_w, rotated_h, lv_draw_buf_width_to_stride(rotated_w,LV_COLOR_FORMAT_RGB888), lv_draw_buf_width_to_stride(rotated_h,LV_COLOR_FORMAT_RGB888), lv_display_get_rotation(disp_drv), LV_COLOR_FORMAT_RGB888);

        for(y=area->y1; y<=area->y2; y++){
            for(x=area->x1; x<=area->x2*3; x=x+3){
                Buffer[(y * ((area->x2+1)*3) + x)] = rotated_buf->blue;
                Buffer[(y * ((area->x2+1)*3) + x+1)] = rotated_buf->green;
                Buffer[(y * ((area->x2+1)*3) + x+2)] = rotated_buf->red;
                rotated_buf++;
            }
        }
        // Free the allocated buffer if needed (depends on your allocation strategy)
         lv_free(rotated_buf);

        // Inform LVGL that the flush is done
        lv_disp_flush_ready(disp_drv);
}

liamHowatt, 2024-08-17:
Just to clarify, you are not using `LV_USE_TFT_ESPI` and `lv_tft_espi_create`? Sorry for giving an answer tailored towards a Arduino+TFT platorm. I got mixed up with the previous discussion. Your platform is not Arduino+TFT.

Have you tried running it yet? What was the result? I think your flush callback could be improved some more. 

Do you know what the stack size is? `rotated_buf` is a very large array that is a local variable so a stack overflow is possible, unless your stack size is 2.5 MB or more. If it's working, then I guess it's fine.

You should not use `lv_free` since `rotated_buf` was not allocated by an allocator.

`rotated_w` and `rotated_h` are used to size the `rotated_buf` before they're declared assigned values. The buffer should be declared after the values are assigned. I'm not sure whether it will compile the way it is. Just so you know variably sized arrays (VLAs) are much less common than constantly sized arrays, though still valid C.

vishvanathdoijad, 2024-08-19:
I improved my flush_cb(); And its work good but I am getting flicker on some lines. Is this flicker are generated due to double buffering...?

void lvgl_flush_cb(lv_display_t *disp_drv, const lv_area_t *area, lv_color_t * colour_p)
{
int32_t x,y,z;
static uint8_t	first_call = 1;

	if( first_call ) {
	    for(y=area->y1; y<=area->y2; y++){
	    	for(x=area->x1; x<=area->x2*3; x=x+3){
	    		Buffer[(y * ((area->x2+1)*3) + x)] = colour_p->blue;
	    		Buffer[(y * ((area->x2+1)*3) + x+1)] = colour_p->green;
	    		Buffer[(y * ((area->x2+1)*3) + x+2)] = colour_p->red;
	    		colour_p++;
	    	}
	    }
		first_call =  0;
		lv_disp_flush_ready( disp_drv );
	}
	if( lv_disp_flush_is_last( disp_drv ) ) {
	    for(y=area->y1; y<=area->y2; y++){
	    	for(x=area->x1; x<=area->x2*3; x=x+3){
	    		Buffer[(y * ((area->x2+1)*3) + x)] = colour_p->blue;
	    		Buffer[(y * ((area->x2+1)*3) + x+1)] = colour_p->green;
	    		Buffer[(y * ((area->x2+1)*3) + x+2)] = colour_p->red;
	    		colour_p++;
	    	}
	    }
	        lv_disp_flush_ready( disp_drv );
	} else
	{
		lv_disp_flush_ready( disp_drv );
		xil_printf("Flush ready for next frame\n\r");
	}
}

vishvanathdoijad, 2024-08-20:
@liamHowatt 
Can you help me to how I can stop the "buffer switching" in flush_cb(); function. Because of that I am getting flicker on screen.

liamHowatt, 2024-08-21:
Looks good!

You are using v8. That's important to know.

I think the first call part might be redundant, yet harmless.

I don't think double buffering would be responsible for the flickering. It might be that you're using partial mode. In your display driver setup, enable `direct_mode` on the `disp_drv` struct. Read https://docs.lvgl.io/8.4/porting/display.html#direct-mode

Could you send a short video so I can see what kind of flickering it is? There are a few different reasons flickering could be happening.