lvgl/lvgl #6214: ESP32-3248S035C with GT911 touch chip, arduino demo compile error

eagl1, 2024-05-11:
### LVGL version

v9.0.0

### What happened?

I ran an example came in `3.5inch_ESP32-3248S035` folder, I don't remember where I downloaded it. It includes `lvgl` and `TFT_eSPI` libraries.

I updated `lvgl` library and still get this error:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

LVGL_Arduino-3.5CTP-gt911:5:10: fatal error: demos/lv_demos.h: No such file or directory

 #include <demos/lv_demos.h>

          ^~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1

demos/lv_demos.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

In the attachment the example file I used:

[LVGL_Arduino-3.5CTP-gt911.zip](https://github.com/lvgl/lvgl/files/15282030/LVGL_Arduino-3.5CTP-gt911.zip)

hape65, 2024-05-11:
you have to copy the demos and the examples folder from ..llibraries/vgl/demos to ..libraries/lvgl/src/demos same for examples

eagl1, 2024-05-11:
I did that:

[image]

And get this error:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

E:\Programs_Files\Arduino\libraries\lvgl-master\src\demos\widgets\assets\img_clothes.c:6:10: fatal error: lvgl/lvgl.h: No such file or directory

 #include "lvgl/lvgl.h"

          ^~~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

eagl1, 2024-05-12:
I just want to include this file as a start:
`E:\Programs_Files\Arduino\libraries\lvgl\src\indev\lv_indev.h`

In the arduino code I did this:
`#include "lvgl/src/indev/lv_indev.h"` but it didn't work. What to do ?

Neothai, 2024-05-12:
> I updated `lvgl` library and still get this error:
>
> ```
> Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz [...]

Have you set `#define LV_USE_DEMO_WIDGETS` in `lv_conf.h` to 1?
##
``` cpp
// Your code
lv_disp_draw_buf_init( &draw_buf, buf, NULL, screenWidth * screenHeight / 6 );
  
/*初始化显示*/
static lv_disp_drv_t disp_drv;
lv_disp_drv_init( &disp_drv );

/*将以下行更改为显示分辨率*/
disp_drv.hor_res = screenWidth;
disp_drv.ver_res = screenHeight;
disp_drv.flush_cb = my_disp_flush;
disp_drv.draw_buf = &draw_buf;
lv_disp_drv_register( &disp_drv );
  
/*初始化（虚拟）输入设备驱动程序*/
static lv_indev_drv_t indev_drv;
lv_indev_drv_init( &indev_drv );
indev_drv.type = LV_INDEV_TYPE_POINTER;
indev_drv.read_cb = my_touchpad_read;
lv_indev_drv_register( &indev_drv );
```
I think the style of your code in this section is probably that of LVGL version 8..
LVGL version 9 The code will look something like this.
``` cpp
/*LVGL draw into this buffer, 1/10 screen size usually works well. The size is in bytes*/
#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))
uint32_t draw_buf[DRAW_BUF_SIZE / 4];

/* LVGL calls it when a rendered image needs to copied to the display*/
void my_disp_flush( lv_display_t *disp, const lv_area_t *area, uint8_t * px_map){
    /*For example ("my_..." functions needs to be implemented by you)
    uint32_t w = lv_area_get_width(area);
    uint32_t h = lv_area_get_height(area);

    my_set_window(area->x1, area->y1, w, h);
    my_draw_bitmaps(px_map, w * h);
     */

    lv_disp_flush_ready(disp); /*Call it to tell LVGL you are ready*/
}

/*Read the touchpad*/
void my_touchpad_read( lv_indev_t * indev, lv_indev_data_t * data ){
    /*For example  ("my_..." functions needs to be implemented by you)
    int32_t x, y;
    bool touched = my_get_touch( &x, &y );

    if(!touched) {
        data->state = LV_INDEV_STATE_RELEASED;
    } else {
        data->state = LV_INDEV_STATE_PRESSED;

        data->point.x = x;
        data->point.y = y;
    }
     */
}

void lvSystemSetup(){
    lv_init();

    disp = lv_display_create(TFT_HOR_RES, TFT_VER_RES);
    lv_display_set_flush_cb(disp, my_disp_flush);
    lv_display_set_buffers(disp, draw_buf, NULL, sizeof(draw_buf), LV_DISPLAY_RENDER_MODE_PARTIAL);

    /*Initialize the input device driver*/
    lv_indev_t * indev = lv_indev_create();
    lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER); /*Touchpad should have POINTER type*/
    lv_indev_set_read_cb(indev, my_touchpad_read);
}
```
##
I think this link might be useful. https://docs.lvgl.io/master/integration/framework/arduino.html

eagl1, 2024-05-13:
> Have you set `#define LV_USE_DEMO_WIDGETS` in `lv_conf.h` to 1?

Yes. But when I use the old lvgl that came with the display folder from the seller, I don't get any error. I've tried to troubleshoot the source of the error when using the new lvgl library but it will be a difficult one to solve in a short time since I don't have a general understanding of how the library folder works.

> I think this link might be useful. https://docs.lvgl.io/master/integration/framework/arduino.html

I did that I got this error:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

E:\Programs_Files\Arduino\libraries\lvgl\src\demos\widgets\assets\img_clothes.c:6:10: fatal error: lvgl/lvgl.h: No such file or directory

 #include "lvgl/lvgl.h"

          ^~~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Neothai, 2024-05-13:
> I did that I got this error:
>
> ```
> Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB [...]
In your `..\lvgl\src\demos\widgets\assets\img_clothes.c` file 
try changing `#include "lvgl/lvgl.h"` to `#include "../../../lvgl.h"` and try compiling again.

eagl1, 2024-05-13:
I did and got this list of errors:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

LVGL_Arduino-3.5CTP-gt911_2:652:21: error: variable or field 'my_disp_flush' declared void

 void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )

                     ^~~~~~~~~~~~~

LVGL_Arduino-3.5CTP-gt911_2:652:21: error: 'lv_disp_drv_t' was not declared in this scope

E:\projects_electronics_programming\tft\esp32_3248s035\LVGL_Arduino-3.5CTP-gt911_2\LVGL_Arduino-3.5CTP-gt911_2.ino:652:21: note: suggested alternative: 'lv_fs_drv_t'

 void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )

                     ^~~~~~~~~~~~~

                     lv_fs_drv_t

LVGL_Arduino-3.5CTP-gt911_2:652:36: error: 'disp' was not declared in this scope

 void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )

                                    ^~~~

E:\projects_electronics_programming\tft\esp32_3248s035\LVGL_Arduino-3.5CTP-gt911_2\LVGL_Arduino-3.5CTP-gt911_2.ino:652:36: note: suggested alternative: 'dup'

 void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )

                                    ^~~~

                                    dup

LVGL_Arduino-3.5CTP-gt911_2:652:42: error: expected primary-expression before 'const'

 void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )

                                          ^~~~~

LVGL_Arduino-3.5CTP-gt911_2:652:76: error: expected primary-expression before '*' token

 void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )

                                                                            ^

LVGL_Arduino-3.5CTP-gt911_2:652:77: error: 'color_p' was not declared in this scope

 void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )

                                                                             ^~~~~~~

E:\projects_electronics_programming\tft\esp32_3248s035\LVGL_Arduino-3.5CTP-gt911_2\LVGL_Arduino-3.5CTP-gt911_2.ino:652:77: note: suggested alternative: 'lv_color_t'

 void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )

                                                                             ^~~~~~~

                                                                             lv_color_t

LVGL_Arduino-3.5CTP-gt911_2:490:24: error: 'TFT_HOR_RES' was not declared in this scope

 #define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))

                        ^~~~~~~~~~~

E:\projects_electronics_programming\tft\esp32_3248s035\LVGL_Arduino-3.5CTP-gt911_2\LVGL_Arduino-3.5CTP-gt911_2.ino:490:24: note: in definition of macro 'DRAW_BUF_SIZE'

 #define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))
[... 279 lines cut ...]
       lv_demo_widgets();            // OK

       ^~~~~~~~~~~~~~~

E:\projects_electronics_programming\tft\esp32_3248s035\LVGL_Arduino-3.5CTP-gt911_2\LVGL_Arduino-3.5CTP-gt911_2.ino:848:7: note: suggested alternative: 'lv_demos_create'

       lv_demo_widgets();            // OK

       ^~~~~~~~~~~~~~~

       lv_demos_create

exit status 1

variable or field 'my_disp_flush' declared void

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

This is the code example I'm using:

[LVGL_Arduino-3.5CTP-gt911_2.zip](https://github.com/lvgl/lvgl/files/15293373/LVGL_Arduino-3.5CTP-gt911_2.zip)

Neothai, 2024-05-13:
Try compiling this again. [LVGL_Arduino-3.5CTP-gt911_2.zip](https://github.com/lvgl/lvgl/files/15297828/LVGL_Arduino-3.5CTP-gt911_2.zip)

eagl1, 2024-05-14:
I had to uncomment these lines:

```
static const uint16_t screenWidth  = 320;
static const uint16_t screenHeight = 480;
```

And add this:
```
typedef struct {
    void * buf1;
    void * buf2;
    uint32_t size; // Size of each buffer in bytes
} lv_disp_draw_buf_t;
```

At `E:\Programs_Files\Arduino\libraries\lvgl\src\display\lv_display.h`

It compiles but nothing shows on the display.

Neothai, 2024-05-14:
> And add this:
> 
> ```
> typedef struct {
>     void * buf1;
>     void * buf2;
>     uint32_t size; // Size of each buffer in bytes
> } lv_disp_draw_buf_t;
> ```
> 
> At `E:\Programs_Files\Arduino\libraries\lvgl\src\display\lv_display.h`

Why do you want to add this?
You want to customize the display buffer, e.g. Buffer size Number of buffers? If so, you can adjust it here.
```cpp
#define DRAW_BUF_SIZE (screenWidth * screenHeight / 10 * (LV_COLOR_DEPTH / 8))

uint8_t draw_buf[DRAW_BUF_SIZE / 4];

/*If you need to use 2 display buffer, uncomment this line*/
//uint8_t draw_buf2[DRAW_BUF_SIZE / 4];

//=====================================================================//

// in lvSystemSetup()
/*If you need to use 2 display buffer, uncomment this line*/
//lv_display_set_buffers(disp, draw_buf, draw_buf2, sizeof(DRAW_BUF_SIZE), LV_DISPLAY_RENDER_MODE_PARTIAL);
```

And this is the latest file I updated. Try compiling it. [LVGL_Arduino-3.5CTP-gt911_3.zip](https://github.com/lvgl/lvgl/files/15304668/LVGL_Arduino-3.5CTP-gt911_3.zip)

eagl1, 2024-05-14:
> Why do you want to add this?
Because it's not there and if I don't add it in one of the header files, I will get an error. I even didn't find any declaration anywhere about that typedef, I got it by asking chatgpt.

> You want to customize the display buffer, e.g. Buffer size Number of buffers? If so, you can adjust it here.

All I want now is to get the display to work. Even TFT_eSPI examples aren't working. So, now I better know what is the display problem and get it to work with TFT_eSPI examples then run lvgl examples.

eagl1, 2024-05-14:
I opened a new issue in Bodmer TFT_eSPI library:
[https://github.com/Bodmer/TFT_eSPI/issues/3323](url)

I'm waiting for the support because nothing shows on the display running any of the library's examples.

eagl1, 2024-05-15:
I flashed the board with the binary in the folder package and it worked to its original demo code it came with:

[image]

[image]

But programming it with arduino platform doesn't work.

eagl1, 2024-06-04:
Anyway, I was able to program the board with:

`#include <Arduino_GFX_Library.h>`.

So I will try to know the related issues on `lvgl` if it worked with me eventually.

closed 2024-06-04 by eagl1

Patsjemoe, 2024-06-30:
hi, where did you find the original bin code ?

eagl1, 2024-06-30:
There's a way using platformio which you can build the code for the ESP32-3248S035C, then you can check the build files which will be related to ESP32-3248S035C in `.pio` folder.

First, download `esp32-smartdisplay-demo`.
1. Install visula studio code.
2. Install C++ and platformio extensions.
3. Initialize a new folder and name it whatever you want, let say "esp32-smartdisplay-demo".
4. Inside the folder, type in the window path CMD and press enter.
5. Enter this command `git clone --recurse-submodules https://github.com/rzeldent/esp32-smartdisplay-demo.git`.
6. Open this repository in vs code and also on the left bar, select platformio logo, then you'll get this set of programming package:

[image]

7. After the Build process, you can go to the project folder and check the `.pio` folder to find the compiled code for your board.

For more information you can refer to the repository source:
[[https://github.com/rzeldent/esp32-smartdisplay-demo](url)](url)

Patsjemoe, 2024-07-01:
Hi,
Thank you for the suggestion, but am a bit hesitant for platformIO, I am
already struggling with too may platforms...
You mentioned you had the capacitive touch working on both arduino and
platformIO. Which example do you have on arduino ? and was this with GFX
library for arduino ?
regards
Ludo

eagl1, 2024-07-01:
Actually I didin't run the capacitive touch on arduino code until now, but I have the source code if you want it. I got it from LVGL arduino demo. And there's a library for the GT911 on github.

Patsjemoe, 2024-07-01:
Hi,
If you have the source code, I would appreciate it if you can send it.
Regards
Ludo

eagl1, 2024-07-02:
[test_tft_espi_gt911_2.zip](https://github.com/user-attachments/files/16060029/test_tft_espi_gt911_2.zip)

Patsjemoe, 2024-07-02:
Thank you for sending this.
Unfortunately :
1)test_tft_espi_gt911_2 :
     fatal error: demos/lv_demos.h: No such file or directory.
   I had the same error with other files I tried....not solved.
2) GT911.ino : 2 errors : esp32/main.cpp:55: undefined reference to
`setup()' and loop....Solved this by adding a setup and empty loop. But
screen just going  black.
Regards
Ludo

Patsjemoe, 2024-07-02:
Hi,
your GT911.ino seems to be a short version of 34-35esp32S35.ino, with loop
and setup and touch deleted...??
Regards
Ludo

eagl1, 2024-07-02:
Yep, I deleted most of the code, because I wanted to compile the code.

Then I moved to platformio setups.

Anyway, I'm working too to get the working source code to embed it in arduino programming.

Patsjemoe, 2024-07-02:
Hi,
To understand : there is still no example working with GT911 capacitive
touch on arduino, but you are working on this?
Regards
Ludo

eagl1, 2024-09-09:
> To understand : there is still no example working with GT911 capacitive touch on arduino, but you are working on this?
To my experience yes.

Now I got it to work on arduino programming, need a bit of time for the GT911 to get it to work.

eagl1, 2024-09-18:
> Hi, To understand : there is still no example working with GT911 capacitive touch on arduino, but you are working on this? 

I started to work on this library:
[https://github.com/TAMCTec/gt911-arduino](url)

The example is working but I have a small problem with the resolution.

docentYT, 2025-03-12:
Hello,
I have similar screen but 7inch version. I've compiled widgets demo successfully, but nothing is displayed on the screen. There is some problem with allocating display buffer:
```c
#ifdef ESP32
  disp_draw_buf = (lv_color_t *)heap_caps_malloc(sizeof(lv_color_t) * screenWidth *screenHeight/4  , MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
#else
  disp_draw_buf = (lv_color_t *)malloc(sizeof(lv_color_t) * screenWidth *screenHeight/4);
#endif
  if (!disp_draw_buf)
  {
    Serial.println("LVGL disp_draw_buf allocate failed!");
  }
  else
  {
  // ...
```

docentYT, 2025-03-12:
I've manually set `screenWidth` and `screenHeight` to smaller value than native resolution and it worked. So this device does not have enough memory, what is strange because it comes with a working demo from the factory.

[image]