lvgl/lvgl #6216: have the C++ version image convert tool and font generate tool?

xinsuinizhuan, 2024-05-13:
### Introduce the problem

I want genetate the image by c++ in my project, not lvgl is c++ and image convert is python

XuNeo, 2024-05-13:
You are very welcome to contribute a C++ version. It could be much faster than python version.
In the mean time, you may want to check the rust version image converter: https://lib.rs/crates/icu_tool

kisvegabor, 2024-05-15:
I'm not sure officially it's worth maintaining a C++ converter too. 

What is your use case for the C++ converter?

xinsuinizhuan, 2024-05-16:
My project is C++， and i want add a picture, then auto convert to a binary file, so my project need add the python code, and configuration python enviroment. If the lvgl is python code, the converter tool also is python, but the lvgl i c++, the the tool is python, is't discordant.

xinsuinizhuan, 2024-05-17:
also font generate tools, with c++

xinsuinizhuan, 2024-05-20:
Some one get the tool, but is for lvlg 8.0

xinsuinizhuan, 2024-05-20:
but this image format how to match in lvlg 9.0?
 {{LV_IMG_CF_TRUE_COLOR, "LV_IMG_CF_TRUE_COLOR"},
                                                                     {LV_IMG_CF_TRUE_COLOR_ALPHA, "LV_IMG_CF_TRUE_COLOR_ALPHA"},
                                                                     {LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED, "LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED"},
                                                                     {LV_IMG_CF_INDEXED_1BIT, "LV_IMG_CF_INDEXED_1BIT"},
                                                                     {LV_IMG_CF_INDEXED_2BIT, "LV_IMG_CF_INDEXED_2BIT"},
                                                                     {LV_IMG_CF_INDEXED_4BIT, "LV_IMG_CF_INDEXED_4BIT"},
                                                                     {LV_IMG_CF_INDEXED_8BIT, "LV_IMG_CF_INDEXED_8BIT"},
                                                                     {LV_IMG_CF_ALPHA_1BIT, "LV_IMG_CF_ALPHA_1BIT"},
                                                                     {LV_IMG_CF_ALPHA_2BIT, "LV_IMG_CF_ALPHA_2BIT"},
                                                                     {LV_IMG_CF_ALPHA_4BIT, "LV_IMG_CF_ALPHA_4BIT"},
                                                                     {LV_IMG_CF_ALPHA_8BIT, "LV_IMG_CF_ALPHA_8BIT"},
                                                                    };

kisvegabor, 2024-05-22:
```
LV_IMG_CF_INDEXED_1BIT -> LV_COLOR_FORMAT_I1
LV_IMG_CF_ALPHA_1BIT -> LV_COLOR_FORMAT_A1
LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED -> Removed
LV_IMG_CF_TRUE_COLOR -> LV_IMG_CF_TRUE_COLOR
LV_IMG_CF_TRUE_COLOR_ALPHA -> LV_COLOR_FORMAT_NATIVE_WITH_ALPHA
```

xinsuinizhuan, 2024-05-23:
LV_IMG_CF_TRUE_COLOR -> LV_IMG_CF_TRUE_COLOR？ LV_IMG_CF_TRUE_COLOR is not exit in lvlg9?

Is this right?
LV_IMG_CF_TRUE_COLOR->LV_COLOR_FORMAT_RGB888
LV_IMG_CF_TRUE_COLOR_ALPHA->LV_COLOR_FORMAT_ARGB8888

kisvegabor, 2024-05-23:
Sorry, copy paste issue. Yes, true color -> native.

kisvegabor, 2024-06-20:
Closing as not planned.

closed 2024-06-20 by kisvegabor