lvgl/lvgl #6218: Rotate Base object (lv_obj) v9

iSetebos, 2024-05-13:
### Introduce the problem

Do I return a Base object (lv_obj) with all objects on it in version 9? In version 8 I used lv_obj_set_style_transform_angle.

liamHowatt, 2024-05-13:
If I understand correctly, yes, that aspect is the same in version 9 compared to version 8.

https://docs.lvgl.io/master/overview/obj.html

iSetebos, 2024-05-13:
lv_img_set_angle and lv_obj_set_style_transform_angle not work.

liamHowatt, 2024-05-13:
Do you get a compiler error or do they not work as intended at runtime?

iSetebos, 2024-05-13:
Compiles but does not work properly.  More precisely, there is no image.

liamHowatt, 2024-05-13:
Is it a binary image? See #6193. This is a known weakness of binary images.

https://docs.lvgl.io/master/widgets/image.html#transformations

> The transformations require the whole image to be available. Therefore indexed images (`LV_COLOR_FORMAT_I1/2/4/8_...`), alpha only images cannot be transformed. In other words transformations work only on normal (A)RGB or A8 images stored as C array, or if a custom [Image decoder](https://docs.lvgl.io/master/overview/image.html#overview-image-decoder) returns the whole image.

iSetebos, 2024-05-13:
This is a png-decoded image in ARGB8888 format, which is located in the RAM array.

liamHowatt, 2024-05-13:
Interesting. Could you please send some code to reproduce the issue?

iSetebos, 2024-05-14:
Load image:

```
void CreatePngImage(T_PNG_IMAGE *Img, const char *mem, uint16_t size, uint16_t w, uint16_t h)
{
  Img->img = fbg_loadPNGFromMemory(mem, size);
  Img->img->data[2] = 4;
 uint32_t x, y, index = 0, color;
	for (uint32_t i=0; i<((w*h)); i++)
		{
			color = Img->img->data[index]<<16;
			color |= Img->img->data[index+1]<<8;
			color |= Img->img->data[index+2];
			if (color == 0xFF00FF)
			{
				Img->img->data[index] = 0;
				Img->img->data[index+1] = 0;
				Img->img->data[index+2] = 0;
				Img->img->data[index+3] = 0;
			}
			index += 4;
		}
 
  uint32_t *img_buff32 = (uint32_t *)(Img->img)->data;
 
  // swap color R<->B
  unsigned char *img_buff = (unsigned char *)(Img->img)->data;
  unsigned char tmp;
  
  for (uint32_t i=0; i<((w*h)); i++)
  {
    tmp = *img_buff;
    *img_buff     = *(img_buff+2);
    *(img_buff+2) = tmp;
    img_buff += 4;
  }
  //
  //Img->dsc.header.always_zero = 0;
  Img->dsc.header.w = w;
  Img->dsc.header.h = h;
  Img->dsc.header.cf = LV_COLOR_FORMAT_ARGB8888; //LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED; 
  Img->dsc.data = (unsigned char *)(Img->img)->data;  
  Img->dsc.data_size = w * h * LV_COLOR_DEPTH / 8;
}
```

Image display:

```
lv_obj_t * imgArrow;
CreatePngImage(&IMG.Arrow,      &_imageArrow,     392,  43,  44);
imgArrow= lv_img_create(osdDirHome.Panel);  
lv_img_set_src(imgArrow, &IMG.Arrow.dsc);
lv_obj_set_size(imgArrow, 43, 44);
lv_obj_set_pos(imgArrow, 48/2-43/2, 48/2-44/2);
lv_img_set_angle(imgArrow, (180+osdDirHome.Angel*10)%360);
```

kisvegabor, 2024-05-14:
> lv_img_set_angle(imgArrow, (180+osdDirHome.Angel*10)%360);

The angle is in 0.1° unit, so it should be:
```c
lv_img_set_angle(imgArrow, (180+osdDirHome.Angel*10)%3600); //%3600
```

iSetebos, 2024-05-15:
> > lv_img_set_angle(imgArrow, (180+osdDirHome.Angel*10)%360);
> 
> The angle is in 0.1° unit, so it should be:
> 
> ```c
> lv_img_set_angle(imgArrow, (1800+osdDirHome.Angel*10)%3600); //1800 and %3600
> ```

I agree.  But that's not the problem.

iSetebos, 2024-05-15:
If the angle is 0, then everything works, in other cases the program hangs.

liamHowatt, 2024-05-15:
You mentioned the CPU goes to 100%. There's a chance an `LV_ASSERT` failed and the default behavior is to spin forever (`while(1);`). Do you have a way of getting a stacktrace or some other way of finding out where it's failing?

closed 2024-06-15 by lvgl-bot