lvgl/lvgl #6229: LV_EVENT_CLICKED should only fire if released happens over the same object where pressed happend

Viatorus, 2024-05-15:
### LVGL version

9.0.0

### What happened?

If a user tries to scroll although scrolling is not possible, unwanted click events occur at the point where the user started scrolling. Even though, the release position happend at a different object.

Think on a list of buttons which is sometimes scrollable, sometimes not.

P.S. Thank you for already fixing this: https://github.com/lvgl/lvgl/pull/5660

### How to reproduce?

Run the code below.
Press the button, than leave the button area and release the mouse. "Clicked" will is printed in the console.

Simulator: https://sim.lvgl.io/v9.0/micropython/ports/webassembly/index.html

```py

# Initialize

import display_driver
import lvgl as lv

# Create a button with a label

    
def event_cb(e):
        print("Clicked")

scr = lv.obj()
btn = lv.button(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)
btn.add_event_cb(event_cb, lv.EVENT.CLICKED, None)
label = lv.label(btn)
label.set_text('Hello World!')
lv.screen_load(scr)
```

liamHowatt, 2024-05-17:
I confirmed this is still a thing in the latest `master`. 

<details>
  <summary>C equivalent for reference</summary>

  ```c
  static void cl_cb(lv_event_t * e)
  {
      LV_LOG_USER("clicked");
  }
  static void click_scroll(void)
  {
      lv_obj_t * btn = lv_button_create(lv_screen_active());
      lv_obj_align(btn, LV_ALIGN_CENTER, 0, 0);
      lv_obj_add_event_cb(btn, cl_cb, LV_EVENT_CLICKED, NULL);
      lv_label_set_text(lv_label_create(btn), "Hello World!");
  }
  ```
</details>

<details>
  <summary>list of buttons in Micropython</summary>

  ```python
  # Initialize
  
  import display_driver
  import lvgl as lv
  
  # Create a button with a label
  
      
  def make_cb(i):
      return lambda e: print("Clicked", i)
  
  scr = lv.obj()
  
  scr.set_flex_flow(lv.FLEX_FLOW.COLUMN)
  for i in range(20):
      btn = lv.button(scr)
      btn.align(lv.ALIGN.CENTER, 0, 0)
      btn.add_event_cb(make_cb(i), lv.EVENT.CLICKED, None)
      label = lv.label(btn)
      label.set_text(str(i).center(20))
      lv.screen_load(scr)
  ```
</details>

This change in lv_indev.c seems to give the desired effect. On release, if we did not scroll, and we're still over the object, send the click.

@kisvegabor, is a PR in order?

```diff
diff --git a/src/indev/lv_indev.c b/src/indev/lv_indev.c
index 08ff5be07..e18b81d14 100644
--- a/src/indev/lv_indev.c
+++ b/src/indev/lv_indev.c
@@ -1310,7 +1310,7 @@ static void indev_proc_release(lv_indev_t * indev)
         }
 
         if(is_enabled) {
-            if(scroll_obj == NULL) {
+            if(scroll_obj == NULL && lv_obj_hit_test(indev_obj_act, &indev->pointer.act_point)) {
                 if(indev->long_pr_sent == 0) {
                     if(send_event(LV_EVENT_SHORT_CLICKED, indev_act) == LV_RESULT_INVALID) return;
                 }
```

kisvegabor, 2024-05-20:
Hi,

It can be controlled by `LV_OBJ_FLAG_PRESS_LOCK`:
```py
btn.remove_flag(btn.FLAG.PRESS_LOCK)
```

Viatorus, 2024-05-21:
Oh, thank you!

Viatorus, 2025-01-13:
Unfortunately, just removing `LV_OBJ_FLAG_PRESS_LOCK` does not work here in this case.

If `LV_OBJ_FLAG_PRESS_LOCK` is not set, a new clickable object will be found and pressed. On release, a click event happens on this new object. That is not what I want. No new press or click event should happen.

If the user's touch goes outside of the pressed object - no other object should be pressed or clicked. Not a new object and also not the old one.

AndreCostaaa, 2025-01-30:
Not stale, PR is open for this

closed 2025-01-30 by liamHowatt

ithinu, 2025-04-23:
I’m using the latest snapshot. Has the fix already been merged into the main branch?

In my case, pressing a button doesn’t generate the click event immediately. However, releasing the pointer anywhere outside the button still triggers a click. Additionally, once the button is initially pressed, it changes visually and stays in that state even if the pointer (still pressed) is moved away, until the pointer is released.

I’ve written my own driver for the CST816S, but the event logs seem fine. Below is a log from lvgl_touchpad_read(lv_indev_t * indev, lv_indev_data_t * data), showing a sequence of events that causes a small button in the center of a 240x320 screen to be clicked when the pointer is released at the screen’s edge:

PRESSED at (123, 162)
PRESSED at (123, 164)
PRESSED at (123, 165)
PRESSED at (123, 165)
PRESSED at (126, 184)
PRESSED at (135, 205)
PRESSED at (139, 216)
PRESSED at (145, 229)
PRESSED at (157, 253)
PRESSED at (163, 263)
PRESSED at (167, 270)
PRESSED at (171, 276)
PRESSED at (175, 282)
PRESSED at (180, 288)
PRESSED at (184, 294)
RELEASED at (188, 297)

PR #7612 (2025-01-30): fix(indev): skip press event on new object release
Fixes #6229 

This PR prevents triggering a press event when the device moves from the originally pressed object to a different object and is then released on that new object.

Testing C code:
```c
static void cl_cb(lv_event_t *e)
{
	LV_LOG_USER("clicked");
}
static void click_scroll(void)
{
	lv_obj_t *btn = lv_button_create(lv_screen_active());
	lv_obj_align(btn, LV_ALIGN_CENTER, 0, 0);
	lv_obj_add_event_cb(btn, cl_cb, LV_EVENT_CLICKED, NULL);
	lv_label_set_text(lv_label_create(btn), "Press This One");
	lv_obj_remove_flag(btn, LV_OBJ_FLAG_PRE [... cut]