lvgl/lvgl #6232: Default fonts not clear on red background

keerthansys22, 2024-05-16:
### LVGL version

v8.3

### What happened?

I’m using default lvgl fonts - montserrat in my screens. While displaying white fonts on red background, there is no clarity and readability. But if the same white fonts displayed on green background, it is displaying perfectly fine. 

I have tried changing the shades of the red background but no major impact on readability.

liamHowatt, 2024-05-17:
Could you please send screenshots/pictures? Or even better, a code snippet to reproduce the issue? If you're using one of the special drawing backends there might be a blending issue with the red color channel specifically. Are you using dave2d, nxp, renesas or any other special drawing backend? Anything else exceptional about your configuration that might be a factor?

[image]
[image]

(on v8.3)

```c
static void red_green_font(void)
{
    lv_obj_set_style_bg_color(lv_scr_act(), lv_palette_main(LV_PALETTE_RED), 0);
    lv_obj_t * lab = lv_label_create(lv_scr_act());
    lv_obj_set_style_text_font(lab, &lv_font_montserrat_48, 0);
    lv_obj_set_style_text_color(lab, lv_color_white(), 0);
    lv_label_set_text(lab, "Lorem Ipsum");
    lv_obj_center(lab);
}
```

keerthansys22, 2024-05-20:
Thanks for your response @liamHowatt, 

[image]

I have not used any special drawing backends for this case. Infact, I'm using the same font on green background and its clear.

keerthansys22, 2024-05-21:
Hi @liamHowatt,

I tried enabling this option in lv_config.h file - /Swap the 2 bytes of RGB565 color. Useful if the display has an 8-bit interface (e.g. SPI)/
#define LV_COLOR_16_SWAP 1

It’s working now and the fonts are more clear and readable. Thanks for your help.

closed 2024-05-21 by keerthansys22

liamHowatt, 2024-05-21:
Nice! :slightly_smiling_face: