lvgl/lvgl #6234: Display is

stevenzhang2024, 2024-05-17:
### LVGL version

v9.1

### What happened?

When we port LVGL9.1. Display like this,is the config error or other problems?
[image]

### How to reproduce?

/*Color depth: 8 (A8), 16 (RGB565), 24 (RGB888), 32 (XRGB8888)*/
#define LV_COLOR_DEPTH 16
/*1: Enable system monitor component*/
#define LV_USE_SYSMON   1
#define LV_USE_OS   LV_OS_FREERTOS

dennisye963, 2024-05-17:
It seems like you are missing color depth conversion, something like "convert_color_depth" function in lvgl/src/extra/libs/png/lv_png.c

stevenzhang2024, 2024-05-17:
Oh ,no. The problem is stm32h7xx cache.  `SCB_EnableICache();SCB_EnableDCache();`

closed 2024-05-17 by stevenzhang2024