lvgl/lvgl #6238: STM32 Baremetal Example does not work on LVGL 9

zacck, 2024-05-18:
### LVGL version

9.1.1-dev

### What happened?

I am trying to use LVGL to render on a 320 by 240 ILI9341 display, my baremetal drivers are working in that I can change the screen's background and I can render  a rectangular shape in a certain color. Please see the little demo video attached. 

MY issue however is that even when I get the project building without errors I can't seem to get any rendering results from LVGL, even though I have supplied the callback and buffers as needed. I am not sure whether I am the one messing up or whether  the instructions are not correct. The instructions on writing the callback are also different some call for using a `uint8_t` whereas another set of instructions call for `lv_color_t` which has me rather confused. 

here is my callback function 

```c
/*
 * Callback used by LVGL to render to the Screen
 */
void lcd_flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * cp){

	//set the drawing region
	bsp_lcd_set_display_area(area->x1, area->y1, area->x2, area->y2);

	int height = area->y2 - area->y1 + 1;
	int width = area->x2 - area->x1 + 1;

	for (int i = 0; i < width * height; i++) {

		bsp_lcd_write(cp, 2);
		cp++;
	}

	//Tell LVGL we are ready to flush
	bsp_lcd_flush();
	lv_display_flush_ready(disp);
}

```

Here's how my  write function looks like 

```c
void bsp_lcd_write(uint8_t *buffer, uint32_t nbytes)
{
	uint16_t *buff_ptr;

	__disable_spi();
	__spi_set_dff_16bit();
	__enable_spi();

	LCD_CSX_LOW();

	buff_ptr = (uint16_t*)buffer;
	while(nbytes){
		while(!REG_READ_BIT(SPI->SR,SPI_SR_TXE_Pos));
		REG_WRITE(SPI->DR,*buff_ptr);
		++buff_ptr;
		nbytes -= 2;
	}

	__disable_spi();
	LCD_CSX_HIGH();
	__spi_set_dff_8bit();
	__enable_spi();

}
```

And as well as a project reproducing my issue. 
https://github.com/zacck/Ili9341_basic_driver

And a video showing what I am seeing 

https://github.com/lvgl/lvgl/assets/108482053/8f125968-7ad5-45db-8653-26ae358e0388

Basically  I can draw with baremetal but LVGL does not

zacck, 2024-05-20:
Update, I managed to get LVGL to work! however not completely as we expect. So now It does render but somehow my code drops a part of the screen.

Here is how I am initializing the screen now 

```c
/*Function to initialize the screen as a LVGL element*/
void screen_init(void){
	lv_color_t *draw_buf1;
	lv_color_t *draw_buf2;

	bsp_lcd_init();

	draw_buf1 = (lv_color_t*)bsp_lcd_get_draw_buffer1_addr();
	draw_buf2 = (lv_color_t*)bsp_lcd_get_draw_buffer2_addr();

	//Init LVGL
	lv_init();

	//Init LVGL with our screen size
	lv_display_t * disp = lv_display_create(BSP_LCD_ACTIVE_HEIGHT, BSP_LCD_ACTIVE_WIDTH);

	//Initialize and set a buffer
	lv_display_set_buffers(disp, draw_buf1, draw_buf2, ((10UL * 1024UL) / 2), LV_DISPLAY_RENDER_MODE_PARTIAL);

	//Set a flush callback so lvgl can draw to our screen
	lv_display_set_flush_cb(disp, lcd_flush_cb);

	lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90);

	lv_display_set_color_format(disp, LV_COLOR_FORMAT_RGB565);
}
```
Amazing I see elements on the screen but part of it is cut out see below.

https://github.com/lvgl/lvgl/assets/108482053/944da449-c833-494d-bb79-29cc2f9a7d34

As you can see my base driver can use the whole screen but now when I hand over to LVGL it uses only a part of the screen, any help would be much appreciated.

kisvegabor, 2024-05-22:
The clipping issue happens because you call `lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90);`. With this LVGL will swap the horizontal and vertical resolution so you also need to configure your display to this orientation.

In ILI9341 it can be set with the ADDRESS_MODE (0x36) command.

zacck, 2024-05-22:
1. Thank you so much for the time! 

I'll give that a bash!

zacck, 2024-05-22:
The solution above fixes this issue thank you @kisvegabor 💯

closed 2024-05-22 by zacck

kisvegabor, 2024-05-26:
Great! :slightly_smiling_face:

zacck, 2024-11-12:
Hey @kisvegabor sorry to contact you this way but I realize it is the one of the only reliable ways  to get to you. We have a community and quite a few of our members are building display based systems. Would you perhaps know of a few power users of LVGL that could come speak and share samples perhaps every other month. Kind Regards and sorry for hacking github into a messaging service.

kisvegabor, 2024-11-13:
Hi @zacck,

No worries at all. :slightly_smiling_face: 

It definitely sounds interesting and I was already thinking about something like this too. These are the topics that I had in mind:

## The true power of LVGL - Features you might not know but you should
Tutorial series, share on LinkedIn and blog

* Software rendered Gradients 
* Observer pattern 
* Transformation and layers 
* Layouts and relative units for responsiveness
* Buffering and rendering modes
* L8, I1 rendering
* Masking (generation and usage)
* Scalability: MCU, MPU, desktop, web
* Hooked drawing
* Profiling 
* Vectors: custom drawing, Lottie, SVG? 
* Matrix

It could be a live coding YouTube cast and then write a blog post/docs page from it.

Unfortunately we are a little bit short on manpower, but I hope next year it will be a little bit better and we will have time for it. Anyway, thank you for bringing it up. It's a great confirmation that not only I think that it could be useful. :slightly_smiling_face:

kisvegabor, 2024-11-17:
I'm not sure. As all these are documented already, the items on the list are rather complex tutorials. However it overlaps a little with examples and demos.

vwheeler63, 2024-11-17:
@kisvegabor Makes sense.  :-)

zacck, 2024-11-23:
Hey @kisvegabor  can I maybe get your email or someway to try and get a meeting with you. Our Lead wants to set out some time to make content that will align with the list above.

kisvegabor, 2024-11-27:
Sounds great! Please send an email to gabor@lvgl.io