lvgl/lvgl #6239: Change TileView Scroll animation speed

jgauchia, 2024-05-18:
### LVGL version

v9.1.0

### What happened?

As described in this [post](https://forum.lvgl.io/t/scroll-animation-speed-in-tile-view-widget/16361) scroll animation speed of a tileview is fixed by "define" constant in **lv_obj_scroll.c**

```
#define SCROLL_ANIM_TIME_MIN    200    /*ms*/
#define SCROLL_ANIM_TIME_MAX    400    /*ms*/
``` 

There are any "possibility" to add a public method or similar in order to change the scroll animation time?

Thank you!

jgauchia, 2024-05-20:
> Hi,
> 
> You can do this to modify the scroll animation:
> ```c
> 
> void anim_start_event_cb(lv_event_t * e)
> {
> 	lv_anim_t * a = lv_event_get_param(e);
> 	if(a) a->duration = 2000;
> }
> 
> ...
> 
>     lv_obj_t * tv = lv_tileview_create(lv_screen_active());
>     lv_obj_add_event_cb(tv, anim_start_event_cb, LV_EVENT_SCROLL_BEGIN, NULL);
> ```

Hi!

Thank you!

kisvegabor, 2024-05-22:
You are welcome! 

I'm closing this issue.

closed 2024-05-22 by kisvegabor