lvgl/lvgl #6246: Bengali string display abnormal using freetype lib

xuanmumy, 2024-05-21:
### LVGL version

v9.1

### What happened?

hello my friend,
I am testing the Bengali language display, but the display results using lvgl+freetype are different from those in Excel. The font file I used is the same.
I tried lvgl8.3+freetype 2.11/2.13 simulator and lvgl9.1+freetype 2.13 simulator,  the results were the same.
excel display:
[image]
freetype display:
[image]
I am a beginner in text display and I want to know if these differences are due to poor support for bengali or there are some libraries not enabled. I see that freetype version 2.7 already supports Bengali.
the string I used here is "মিউজিক প্লেয়ার" and I converted it into a char array to use in lvgl ui . font file is attached.
[Bengali.zip](https://github.com/lvgl/lvgl/files/15384016/Bengali.zip)

could you give me some advice?
thanks!

### How to reproduce?

```
char bengali_list[250] = { 0xe0, 0xa6, 0xae, 0xe0, 0xa6, 0xbf, 0xe0, 0xa6, 0x89, 0xe0, 0xa6, 0x9c, 0xe0, 
0xa6, 0xbf, 0xe0, 0xa6, 0x95, 0x20, 0xe0, 0xa6, 0xaa, 0xe0, 0xa7, 0x8d, 0xe0, 0xa6, 
0xb2, 0xe0, 0xa7, 0x87, 0xe0, 0xa7, 0x9f, 0xe0, 0xa6, 0xbe, 0xe0, 0xa6, 0xb0, 0x0 };
    static lv_ft_info_t info;
    info.name = "./lvgl/examples/libs/freetype/Bengali.ttf";
    info.weight = 24;
    info.style = FT_FONT_STYLE_NORMAL;
    info.mem = NULL;
    if(!lv_ft_font_init(&info)) {
        LV_LOG_ERROR("create failed.");
    }
    static lv_style_t style;
    lv_style_init(&style);
    lv_style_set_text_font(&style, info.font);
   lv_obj_t * label = lv_label_create(lv_scr_act());
   lv_obj_add_style(label, &style, 0);
   lv_label_set_text(label, bengali_list);
```

kisvegabor, 2024-05-26:
Hi,

Probably it's due to the poor Bengali support. :slightly_frowning_face: 

I myself don't know the details of the Bengali writing system. I've asked the latest ChatGPT-4o, but it couldn't help much. 

So far we haven't needed a real text shaping, but for Indian scripts we should have one. It means two things:
1. Implement text shaping in general
2. Make it support Bengali

Unfortunately, these are not on my roadmap now, but if someone is interested in working on these, I would be ready to help. However, it would be a very deep rework on LVGL's text processing.

xuanmumy, 2024-05-26:
thanks for your reply.

xlbao123, 2024-06-05:
It needs a text shaping engine, like harfbuzz.

closed 2024-06-30 by lvgl-bot