lvgl/lvgl #6247: Cannot play custom gif in v9.1

hoanglam9925, 2024-05-21:
### LVGL version

v9.1.0

### What happened?

I can run the example gif without any issues by using lv_example_gif_1(). If I add this code along with the necessary files to an Arduino project, it still functions, but if I change the gif. Using this link: https://lvgl.io/tools/imageconverter_v9, I attempted to convert a gif, but it did not work. I tried all the color formats. 
I convert using bulb.gif found in the example folder.

### How to reproduce?

void lv_gif(void)
{
    LV_IMAGE_DECLARE(bulb);
    lv_obj_t * img;

    img = lv_gif_create(lv_screen_active());
    lv_gif_set_src(img, &bulb);
    lv_obj_align(img, LV_ALIGN_LEFT_MID, 20, 0);
}

liamHowatt, 2024-05-24:
Hey, thanks. Indeed the v9 tool doesn't work. The [docs](https://docs.lvgl.io/master/libs/gif.html) still link to the v8 converter but the output is not compatible with v9 LVGL. We'll keep this issue open until it's corrected.

Until then, 2 thing worked for me.

### 1.

Use the v8 converter and then edit the `lv_img_dsc_t` at the bottom of the output.

E.g. change this output from the v8 converter...
```c
const lv_img_dsc_t bulb = {
  .header.cf = LV_IMG_CF_RAW_CHROMA_KEYED,
  .header.always_zero = 0,
  .header.reserved = 0,
  .header.w = 60,
  .header.h = 80,
  .data_size = 17693,
  .data = bulb_map,
};
```
...to this v9 compatible one...
```c
const lv_img_dsc_t bulb = {
  .header.cf = LV_COLOR_FORMAT_RAW,
  .header.w = 60,
  .header.h = 80,
  .data_size = 17693,
  .data = bulb_map,
};
```

### 2.

Or you can use the python tool. It worked for me.

```
python  scripts/LVGLImage.py  --ofmt C  --cf RAW  -o your_output_dir_path   examples/libs/gif/bulb.gif
```

hoanglam9925, 2024-05-24:
The first method worked for me, but it only worked with images smaller than 140 x 140. When I tried to use a file 150 x 150, there were some issues; in the log, it simply said "Couldn't load the source" for images larger than this. Even after raising LV_MEM_SIZE to 128 * 1024, it is still not functioning. How to play a larger-sized GIF
And I have tried play gif from FATFS, but always crash. This is my code:
[image]
I already enable LV_USE_FS_FATFS flag in config file
[image]
how to play gif from fatfs. i flashed a .gif file to fatfs

hoanglam9925, 2024-05-26:
LVGLImage.py allowed me to try converting a gif to.C, and it worked. However, when I tried to convert to a.bin file to play from a file system, it consistently returned a.c file. I verified that the function in LVGLImage.py forces a.C file when the --cf option is set to RAW or RAW_ALPHA. 
I tried to convert a .png file to .bin file it worked smoothly.
How i can to convert a gif to .bin file?

liamHowatt, 2024-05-29:
I was able to load a 480x360 gif from a C file and from the FS as a `.gif`. I don't know if 128kb is enough to load your gif and satisfy the other LVGL memory needs. The memory requirements of the GIF decoder are in the docs: https://docs.lvgl.io/master/libs/gif.html#memory-requirements so if you're using 16 bit depth mode, the decoder will use 90 kB, which is almost all of the 128 kB.

Yes I see what you mean. cc @XuNeo. I think it's because there's no use case for bin files in raw format. The GIF decoder operates on C arrays or unmodified .gif files, not .bin files.

XuNeo, 2024-05-29:
>  The GIF decoder operates on C arrays or unmodified .gif files, not .bin files.

Yes. There's no need to (and cannot ) change the suffix from `.gif` to `.bin`

hoanglam9925, 2024-05-29:
I retested, and this time it worked when I loaded a.gif file from fatfs using the cloned library on May 27. The library I cloned on May 11th still throws errors.
[image]
Between two versions, is there a new version?

liamHowatt, 2024-05-30:
I tested just now, I got the same behavior on both. But there could be something that changed which affects your platform :slightly_smiling_face:

I think you're almost out of memory and you do run out of memory in the May 11th version due to some subtle change in allocation patterns between versions. Recall this from my comment:

> I don't know if 128kb is enough to load your gif and satisfy the other LVGL memory needs. The memory requirements of the GIF decoder are in the docs: [...]

With that being said, [here are the changed files](https://github.com/lvgl/lvgl/compare/b16aa3c55...master) since May 11th. That view is not very useful but I went through the list and the most relevant thing that changed related to GIF+FATFS was the lv_fs caching system was refactored and write caching was added. You're not writing though, so I'm not sure.

To clarify, it works with the latest version but not the May 11th version? The one thing I can think of is to try setting `LV_FS_FATFS_CACHE_SIZE` to `0` if it isn't already, and see if that makes the May 11th version work for you.

hoanglam9925, 2024-06-03:
LV_FS_FATFS_CACHE_SIZE is always 0. Even after increasing LV_MEM_SIZE to 256*1024, the error message still shows. I believe that something related to memory was improved in the most recent version, and it works.

liamHowatt, 2024-06-05:
256 kB sounds like enough! Since it's fixed, it's a less critical issue. If you're using Arduino IDE 2, I believe there's interactive debugging. You're welcome to try to pinpoint the cause of the failure and we can improve the related code if there's a risk of it still happening in the new versions.

kisvegabor, 2024-06-20:
I close this issue because there was no activity here for a while.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2024-06-20 by kisvegabor

Satnet, 2024-11-08:
Indeed the LVGL Image Converter does not work if you use the V9 version. Using V8 and then manually changing to LV_COLOR_FORMAT_RAW, does work fine.

Also loading from SD Card has many issues:
* JPG works
* BMP shows up like garbage on screen and instant crash
* PNG dont work LVGL reports it can't load it, I printed the file size and it shows 0, but I verified that the file is loadable from the 
file system, I even printed the first 10 bytes of the file at one point.
```
Opening file: /image.png
File opened successfully.
Opening file: /image.png
File opened successfully.
Opening file: /image.png
File opened successfully.
Opening file: /image.png
File opened successfully.
Opening file: /image.png
File opened successfully.
[Warn]	(6.284, +6284)	 decoder_open: Failed to load PNG file: error 78: failed to open file for reading, data size: 0
 lv_lodepng.c:154
```

* GIF (also from SD as the images) shows up as a solid color slows down the device and crashes eventually.
LVGL reports here as well that it can't open the file.
```
Opening file: /image.jpg
File opened successfully.
Opening file: /image.jpg
File opened successfully.
Attempting to load GIF from: S:/gif.gif
Opening file: /gif.gif
Error opening file!
Failed to open GIF file for test read.
[Info]	(6.270, +104)	 lv_obj_create: begin lv_obj.c:215

[Info]	(6.270, +0)	 lv_label_create: begin lv_label.c:125

[Info]	(6.270, +0)	 lv_label_create: begin lv_label.c:125

[Info]	(6.271, +1)	 lv_label_create: begin lv_label.c:125

[Info]	(6.271, +0)	 lv_label_create: begin lv_label.c:125

[Info]	(6.271, +0)	 lv_label_create: begin lv_label.c:125
```

GIF as array works as mentioned above, if converting using the LVGL 8 converter.

My device has SDRAM of 32 megabytes, I've allocated a massive 10 megabytes to the LVGL buffer which is placed in SDRAM. So there's plenty of space.

Satnet, 2024-11-11:
> Is this with the online converter or LVGLImage.py?
> 
> For your size-zero loading issue, which FS abstraction are you using? Also, in that output, it looks like it works a few times with the same image and then fails? Is there a memory leak?

I am not sure if this question was for me or not but If it was, here is my reply.

This is for the online converter, I have not tried the scripts yet, I am going to try it.