lvgl/lvgl #6251: lvgl draws artefacts at simple button click

Wlad79, 2024-05-22:
### LVGL version

v8.4

### What happened?

I have followed the instruction at https://docs.lvgl.io/8.4/get-started/quick-overview.html#add-lvgl-into-your-project and used first example at https://docs.lvgl.io/8.4/examples.html. And got followed result: https://github.com/lvgl/lvgl/assets/54247746/22e8ab13-f8eb-4ee6-b354-b1fd21fb186e . how can the artefacts be eliminated?

### How to reproduce?

My Barmetal is STM32F7 with SDRAM (uVision / Keil) and followed Code in "void my_disp_flush(lv_disp_drv_t * disp, const lv_area_t * area, lv_color_t * color_p)":

```
for(y_pos = area->y1; y_pos <= area->y2; y_pos++) {
  for(x_pos = area->x1; x_pos <= area->x2; x_pos++) {
    uint32_t temp = (0xC0000000 + (4*(y_pos*(LCD_X_SIZE) + x_pos)));
    *(uint32_t *) temp = (uint32_t)(0xff<<24 | color_p->ch.red<<16 | color_p->ch.green<<8 | color_p->ch.blue);
    color_p++;
  }
}
```

liamHowatt, 2024-05-24:
Hey. I think I spotted the problem. Your display's physical framebuffer base address is `0xC0000000`. The issue is that you're also using that address for the draw buffer. You should use a different memory area for the draw buffer.

The line where you call `lv_disp_draw_buf_init` and pass `buf1` (a pointer to `0xC0000000`) is telling LVGL that it can use that memory address as a place where it can temporarily buffer pixels before they get flushed to the real display. That's why there's a distorted thin blue bar at the top.

Can you reserve some different memory for the draw buffer?

Here I'm passing a new static array buffer to use instead of `0xC0000000`.
```c
static uint8_t draw_buf_mem_buf[LCD_X_SIZE * LCD_Y_SIZE*4+4*LCD_X_SIZE];
lv_disp_draw_buf_init(&draw_buf, draw_buf_mem_buf, NULL, LCD_X_SIZE * LCD_Y_SIZE*4+4*LCD_X_SIZE );  /*Initialize the display buffer.*/
```

I'm not aware of your full STM+uVision+Keil setup and how much external SDRAM you added, but you may need to make the draw buffer smaller than the frame buffer, which is valid in LVGL.

```c
static uint8_t draw_buf_mem_buf[LCD_X_SIZE * LCD_Y_SIZE * 4 / 10];
lv_disp_draw_buf_init(&draw_buf, draw_buf_mem_buf, NULL, LCD_X_SIZE * LCD_Y_SIZE * 4 / 10);  /*Initialize the display buffer.*/
```

You might even be able to use your variable `buf2` if that's unoccupied memory and large enough. That may even be what you meant to do.

```c
lv_disp_draw_buf_init(&draw_buf, buf2, NULL, LCD_X_SIZE * LCD_Y_SIZE*4+4*LCD_X_SIZE );  /*Initialize the display buffer.*/
```

Wlad79, 2024-05-24:
Thanks for the response. It solves the problem with top bar, if the code changes to this:
```
static lv_color32_t *buf1 = (lv_color32_t *)0xC0100000;
lv_disp_draw_buf_init(&draw_buf, buf1, NULL, LCD_X_SIZE * LCD_Y_SIZE );// shows full rendered screen
//lv_disp_draw_buf_init(&draw_buf, buf1, NULL, LCD_X_SIZE * LCD_Y_SIZE *4 /10 );// shows only top bar with rendered content
//lv_disp_draw_buf_init(&draw_buf, buf1, NULL, LCD_X_SIZE * LCD_Y_SIZE /10 );// shows only half screen with rendered content
```
But I run in next Problem with lvgl@STM32F7 timing, maybe you have an advice on that too. Somehow refresh rate is very low, although I'm using LTDC-Driver. Keil uVision says me that there no execution possible in interval from 1 til 10 ms, because many timing error are happen (I'm using EventRecorder from Middleware of uVision). Special I can only have 20 ms with errors from EventRecorder). But the execution of lv_tick_inc(10) takes 103us with 20ms interval: 
[image]

Now the question to that setup: Is it possible, that Program with lv_demo_widgets() frees after a time and lv_example_keyboard_1() has problems with showing pressed character.

liamHowatt, 2024-05-29:
Are you calling `lv_tick_inc(10)` in a timer interrupt? Make sure you're incrementing the tick accurately. If you can't use a timer interrupt, then run this: `lv_tick_inc(HAL_GetTick() - lv_tick_get())` in your main loop every loop iteration. In LVGL v9, you can do `lv_tick_set_cb(HAL_GetTick)`. Are you delaying in the loop `HAL_Delay(10)` or similar? Try removing that, if you are.

> Now the question to that setup: Is it possible, that Program with lv_demo_widgets() frees after a time and lv_example_keyboard_1() has problems with showing pressed character.

If lv_demo_widgets is freezing, try increasing `LV_MEM_SIZE` in lv_conf.h. For they keyboard pressed character issue, send a short video please, if it's still occurring after fixing the timing problems.

Wlad79, 2024-05-29:
thanks so fare. I will post regarding last problems in new issues.

closed 2024-05-29 by Wlad79