lvgl/lvgl #6263: Grid layout style `grid_column_align` seems to have no effect

nagyrobi, 2024-05-24:
### LVGL version

v8.4 and master

### What happened?

All variations of `grid_column_align` - no matter what value I set the columns are always left-aligned.

[image]

Also posted in [forum](https://forum.lvgl.io/t/grid-layout-style-grid-column-align-seems-to-have-no-effect/16535/1), but no reaction...

### How to reproduce?

```c
static const lv_coord_t page_1_row_dsc[] = {LV_GRID_FR(1),LV_GRID_FR(1), LV_GRID_TEMPLATE_LAST};
static const lv_coord_t page_1_column_dsc[] = {LV_GRID_FR(1),LV_GRID_FR(1), LV_GRID_TEMPLATE_LAST};
lv_style_t *bdr_style;
lv_obj_t *page_1;
lv_obj_t *cell_0_0;
lv_obj_t *cell_0_1;
lv_obj_t *cell_1_0;
lv_obj_t *cell_1_1;

void setup_page() {
    bdr_style = malloc(sizeof(lv_style_t));
    lv_style_init(bdr_style);
    lv_style_set_align(bdr_style, LV_ALIGN_CENTER);
    lv_style_set_border_color(bdr_style, lv_color_hex(0x808080));
    lv_style_set_border_width(bdr_style, 2);
    lv_style_set_pad_all(bdr_style, 4);
    page_1 = lv_obj_create(lv_scr_act());
    lv_obj_set_layout(page_1, LV_LAYOUT_GRID );
    lv_obj_set_style_grid_row_dsc_array(page_1, page_1_row_dsc, 0);
    lv_obj_set_style_grid_column_dsc_array(page_1, page_1_column_dsc, 0);
    lv_obj_set_style_grid_column_align(page_1, LV_GRID_ALIGN_CENTER, 0);
    lv_obj_set_style_border_color(page_1, lv_color_hex(0xFF0000), (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_border_width(page_1, 2, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_pad_all(page_1, 10, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_width(page_1, lv_pct(100), (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    cell_0_0 = lv_label_create(page_1);
    lv_obj_add_style(cell_0_0, bdr_style, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_row_pos(cell_0_0, 0, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_column_pos(cell_0_0, 0, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_row_span(cell_0_0, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_column_span(cell_0_0, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_label_set_text(cell_0_0, "Cell 0/0" );
    cell_0_1 = lv_label_create(page_1);
    lv_obj_add_style(cell_0_1, bdr_style, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_row_pos(cell_0_1, 0, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_column_pos(cell_0_1, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_row_span(cell_0_1, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_column_span(cell_0_1, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_label_set_text(cell_0_1, "Cell 0/1" );
    cell_1_0 = lv_label_create(page_1);
    lv_obj_add_style(cell_1_0, bdr_style, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_row_pos(cell_1_0, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_column_pos(cell_1_0, 0, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_row_span(cell_1_0, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_column_span(cell_1_0, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_label_set_text(cell_1_0, "Cell 1/0" );
    cell_1_1 = lv_label_create(page_1);
    lv_obj_add_style(cell_1_1, bdr_style, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_row_pos(cell_1_1, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_column_pos(cell_1_1, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_row_span(cell_1_1, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_obj_set_style_grid_cell_column_span(cell_1_1, 1, (int)LV_STATE_DEFAULT|(int)LV_PART_MAIN);
    lv_label_set_text(cell_1_1, "Cell 1/1" );
}
```

kisvegabor, 2024-05-25:
Szia Robi! :slightly_smiling_face: 

It has no effect because the columns has `fr(1)` width. Due to this they fill the whole width of the parent and they cannot be aligned at all as there is no space to move them.

You can either
1. Not fill the whole width:
```c
static const lv_coord_t page_1_column_dsc[] = {150, 150, LV_GRID_TEMPLATE_LAST};
```
2. Align the cell in the column:
```c
lv_obj_set_style_grid_cell_x_align(cell_1_1, LV_GRID_ALIGN_CENTER, 0);
```
[image]

nagyrobi, 2024-05-25:
Szia!

As we understand from the doc, the whole width of the parent should be treated as `fr(1)` + `fr(1)`, that means, if eg. the parent is 200px wide, the grid cells widths should each result to 100px.

Now if I place smaller children within these cells, I should still be able to commonly set an alignment for them.

So 1. - that's the whole point of `fr(xx)` stuff, to not specify discrete widths, but calculate them on the run, 2. - individual alignments of course work for each cell, but that defeats the purpose of the single line `grid_column_align`

nagyrobi, 2024-05-25:
@clydebarrow

kisvegabor, 2024-05-25:
Aligning the column and the content are two independent things. Imagine a single 100 px wide column. You can align this column to the left, center or right, and separately set where to place the cells in it.

I understand that having something like [`justify-items`](https://css-tricks.com/snippets/css/complete-guide-grid/#aa-justify-items) in CSS might make sense in LVGL too. By default the cell's alignment could be `auto` so the parent's `justify-items` is applied however it can be overwritten individually in each cell.

nagyrobi, 2024-05-25:
That would be more intuitive, yes.

nagyrobi, 2024-05-30:
Pinging my colleague @clydebarrow

closed 2024-06-23 by lvgl-bot