lvgl/lvgl #6270: Buggy Demos in 9.x

alexriegler12, 2024-05-24:
### LVGL version

9.x

### What happened?

Some Demos on the documentation website don't work anymore or have glitches, for examples:
"add fade mask to roller" just displays a white screen
The scale demo with two gauges allows to move the needle of the right gauge that uses an image as needle like it was scrollable. This doesn't seem to be right.

kisvegabor, 2024-05-30:
Hi,

I've fixed the roller example here: https://github.com/lvgl/lvgl/pull/6307

Do you mean this scale example?
[image]

If so, it looks good to me.

alexriegler12, 2024-05-31:
the needle of the right scale can be moved by scrolling to the left or right. this doesn't seem right.

liamHowatt, 2024-06-01:
Good catch. The needle on the second scale is horizontally scrollable. You can drag the needle left and right. PR open: #6320

PR #6307 (2024-06-03): fix(roller): enable lv_example_roller_3 again
fixes #6270

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kc [... cut]
PR #6320 (2024-06-03): fix(example): lv_example_scale_3 second scale needle was scrollable
### Description of the feature or fix

#6270

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kcon [... cut]
closed 2024-06-03 by FASTSHIFT