lvgl/lvgl #6271: tiny_ttf ignores cache_size in lv_tiny_ttf_create_file_ex function

cybekRT, 2024-05-24:
### LVGL version

v9.1.0, master

### What happened?

The documentation ( https://docs.lvgl.io/master/libs/tiny_ttf.html ) states that the default cache size is 4kB and can be modified by calling the function with `_ex` suffix. However in the code ( https://github.com/lvgl/lvgl/blob/master/src/libs/tiny_ttf/lv_tiny_ttf.c#L273 ) the cache_size parameter is unused.

kisvegabor, 2024-05-30:
Hi,

@W-Mai is our cache expert. :slightly_smiling_face: 
Could you take a look at it?

W-Mai, 2024-06-05:
> Hi,
> 
> @W-Mai is our cache expert. 🙂 Could you take a look at it?

Ok, let me take a look in it. 😁

cybekRT, 2024-06-07:
Hi, thank you very much for your effort! I've checked and it looks like the cache is properly created and used now :)
Unfortunately, looks like the stb_truetype_htcw doesn't use cache, so in case when the font is in flash (file api), it still does many uncached reads. But even so, now it is definitely faster than previously, so thanks again for your work!

W-Mai, 2024-06-07:
> Hi, thank you very much for your effort! I've checked and it looks like the cache is properly created and used now :) Unfortunately, looks like the stb_truetype_htcw doesn't use cache, so in case when the font is in flash (file api), it still does many uncached reads. But even so, now it is definitely faster than previously, so thanks again for your work!
> 
> One thing worth noting, the max cache size parameter is applied to two caches, so if user requests limit of 32kB of cache, it will use up to 64kB, 32kB for glyphs and 32kB for bitmap. There is probably no better way to provide one limit for two caches, but maybe worth noting in the documentation or somewhere.

I don't know if you know that lvgl's file system has cache function. You can take a look at this related api.
- LV_FS_XXXX_CACHE_SIZE

cybekRT, 2024-06-07:
> I don't know if you know that lvgl's file system has cache function. You can take a look at this related api.

I know, but if I understand correctly, it caches the linear space of file, and since the ttf library is doing many seeks, the cache size must be huge enough to work. It would be nice to have a ttf tables cached, but I understand that it is huge task and I do not complain. I am just doing some project on the ESP32E without spiram, and I was surprised that rendering short text from ttf is taking like about 30 seconds (it was much more before the caching added to tinyttf :) ).

kisvegabor, 2024-06-24:
Let's continue in https://github.com/lvgl/lvgl/pull/6333

closed 2024-06-24 by kisvegabor