lvgl/lvgl #6279: Play lottie example

hoanglam9925, 2024-05-26:
### LVGL version

v9.1.1

### What happened?

I tried running lv_example_lottie_1, which runs lottie from a .c file, after seeing this library upgrade recently to support play lottie. However, example_2—which runs from a source—showed a blank screen and produced no error log. Does any flag need to be enabled?

kisvegabor, 2024-05-26:
Hi,

Happy that at least it's working at least from an array :slightly_smiling_face: 

Do you try it on PC or an embedded system?

hoanglam9925, 2024-05-26:
I'm running on embedded system. In ESP32-S3 with arduino

kisvegabor, 2024-05-30:
Can you open the file for reading using normal file system functions?

liamHowatt, 2024-06-01:
Lottie file loading uses stdio.h file functions. It doesn't use LVGL FS layer. Is that corret kisvegabor? I believe for loading Lottie files, the platform needs to support those stdio.h file interfaces.

Here is where the lottie file gets opened: https://github.com/lvgl/lvgl/blob/master/src/libs/thorvg/tvgLottieLoader.cpp#L202

And we document it here: https://docs.lvgl.io/master/widgets/lottie.html#set-a-source

> Lottie animations can be opened from JSON files by using lv_lottie_set_src_file(lottie, "path/to/file.json"). Note that the Lottie loader doesn't support LVGL's File System interface but a "normal path" should be used without a driver letter.

Finally, make sure the path you use is either absolute, or relative to the "working directory", whatever that means on ESP32-S3 Arduino. When deployed on-device, the hardcoded path in `lv_example_lottie_2` likely will need to be adjusted.

Does ESP32-S3 on Arduino have a filesystem that can be accessed through `fopen`, `fread`, `fclose`, etc?

hoanglam9925, 2024-06-04:
I'm currently using FATFS. I use the default fsdrv found in lvgl/src/libs/fsdrv in the lvgl lib. I can see that this file already uses f_open from ff.h, but when I try to load a lottie, it just displays a blank screen.
I can use this driver to load a gif.

liamHowatt, 2024-06-04:
Ah, okay. Currently you cannot load Lotties from files with FATFS. Lottie files are not loaded through the LVGL filesystem abstraction so they cannot leverage anything in fsdrv. Lottie files are loaded by logic in lvgl/src/libs/thorvg which directly calls stdio.h functions such as `fopen`. Most everything else in LVGL uses fsdrv. Lottie is the exception.

We could start discuss making Lottie load from fsdrv.

kisvegabor, 2024-06-07:
With @becseya we were talking about adding `LV_FS_DRIVER_FORCE   'A'` config option. This way if a no driver letter is specified it will work with both LVGL file system and without it too. For example:
- `path/to/file` if LVGL FS is not used: use normal C FS functions as they are in the 3rd party library (e.g. in Lottie)
- `path/to/file` if LVGL FS is used: as the path doesn't start with letter + `:` the path will passed to the POSIX driver as it is

becseya, 2024-06-13:
We added a detailed documentation on how to use the filesystem under ESP in #6367 (and also the `DEFAULT_DRIVE_LETTER` mentioned by @kisvegabor )

@hoanglam9925 you should try migrating to [ESP's FATFS](https://github.com/espressif/esp-idf/tree/master/examples/storage/fatfsgen) instead of LVGL's

kisvegabor, 2024-06-20:
If it's still not working please comment in https://github.com/lvgl/lvgl/pull/6367

closed 2024-06-20 by kisvegabor

Yul81, 2025-01-10:
@hoanglam9925 What did you do in orther to make it work? currently when trying to run the lottie1 example my application crashes. I am using a M5Stack Dial - ESP32-S3 Smart Rotary Knob w/ 1.28" Round Touch Screen.
I configured 8MB flash

hoanglam9925, 2025-01-10:
@Yul81 I simply import the example and run it; I don't do anything special. However, the code I used for this example was the most recent commit at the time I create this issue, not the most recent code as of right now.
Instead of downloading this code from the Arduino library, I cloned it and copied it as a library. 
This repository is updated daily, and the Arduino library may be out of date.

Here is my complete code run using example 1:

/*Using LVGL with Arduino requires some extra steps:
 *Be sure to read the docs here: https://docs.lvgl.io/master/get-started/platforms/arduino.html  */

#include <lvgl.h>
#include <TFT_eSPI.h>

#include <examples/lv_examples.h>
//#include <demos/lv_demos.h>

/*Set to your screen resolution*/
#define TFT_HOR_RES   240
#define TFT_VER_RES   240

/*LVGL draw into this buffer, 1/10 screen size usually works well. The size is in bytes*/
#define DRAW_BUF_SIZE (TFT_HOR_RES * TFT_VER_RES / 10 * (LV_COLOR_DEPTH / 8))
void *draw_buf;
unsigned long lastTickMillis = 0;
static lv_display_t * disp;

void setup()
{
    String LVGL_Arduino = "Hello Arduino! ";
    LVGL_Arduino += String('V') + lv_version_major() + "." + lv_version_minor() + "." + lv_version_patch();

    Serial.begin( 115200 );
    Serial.println( LVGL_Arduino );

    lv_init();

    draw_buf = heap_caps_malloc(DRAW_BUF_SIZE, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL);
    disp = lv_tft_espi_create(TFT_HOR_RES, TFT_VER_RES, draw_buf, DRAW_BUF_SIZE);

//    lv_obj_t *label = lv_label_create( lv_scr_act() );
//    lv_label_set_text( label, LVGL_Arduino.c_str());
//    lv_obj_align( label, LV_ALIGN_CENTER, 0, 0 );
//    lv_demo_benchmark();

//    lv_obj_t * lottie = lv_rlottie_create_from_file(lv_screen_active(), 200, 200, "lottie.json");
//    lv_obj_center(lottie);
//    lv_demo_lottie();
    lv_example_rlottie_1();
    

    Serial.println( "Setup done" );
}

void loop()
{
    unsigned int tickPeriod = millis() - lastTickMillis;
    lv_tick_inc(tickPeriod);
    lastTickMillis = millis();
    lv_task_handler(); /* let the GUI do its work */
//    delay(5); /* let this time pass */
}

Yul81, 2025-01-10:
Who calls the setup function? also is this the right example? since you are calling a lv_example_rlottie I thought you were using the lottie widget. I believe maybe is a typo, in any case what is you setup for LV_DRAW_BUF_ALIGN and for LV_DRAW_BUF_STRIDE_ALIGN?

Thanks

hoanglam9925, 2025-01-11:
You can see how an Arduino works by looking at this example, where the setup function runs once and the loop function is the main function.
LV_DRAW_BUF_ALIGN and LV_DRAW_BUF_STRIDE_ALIGN i use default value

liamHowatt, 2025-01-14:
It's important to note that lottie and rlottie are two different widgets in LVGL.

justdying, 2025-04-27:
You should use absolute paths instead of relative paths. I tested it on Windows, and the path must use double backslashes \\
for example:

`lv_lottie_set_src_file(lottie, "D:\word\\xiaozhi-ai\\code\\lv_port_pc_visual_studio\\LvglPlatform\\lvgl\\examples\\widgets\\lottie\\lv_example_lottie_approve.json");`

and it will work

justdying, 2025-04-27:
Oh, there's a missing /. Resubmitting.

`lv_lottie_set_src_file(lottie, "D:\\word\\xiaozhi-ai\\code\\lv_port_pc_visual_studio\\LvglPlatform\\lvgl\\examples\\widgets\\lottie\\lv_example_lottie_approve.json");`