lvgl/lvgl #6280: More color depths

alexriegler12, 2024-05-26:
### Introduce the problem

It seems that some color depths from LVGL 8 are missing in LVGL 9.
LVGL 8 for examples also supports rgb332.

I have a project where the framebuffer is rgba5551.
Currently, I convert the colors in the flush callback by bit shifting, but that is slow.
It would be good for LVGL to support more color depths or even add a modular color depth driver model with callbacks that convert the internal color format to something the display understands (similar to set_px_cb that I believe existed in older versions, but @ better approach)

alexriegler12, 2024-05-27:
A callback Interface similar to indev and display flush would be a good option. That would make the library more modular and better suitable for uncommon hardware configurations (for example it could improve monochrome support)

alexriegler12, 2024-06-18:
I don't have enough knowledge of the internals of LVGL.
Does v9.x Work with monochrome Screens?

kisvegabor, 2024-06-20:
> Does v9.x Work with monochrome Screens?
We are working on it! See  https://github.com/lvgl/lvgl/pull/6345

kisvegabor, 2024-07-10:
Let continue in https://github.com/lvgl/lvgl/pull/6345

closed 2024-07-10 by kisvegabor