lvgl/lvgl #6281: 9.1 Input Device Documentation

AlexBestoso, 2024-05-26:
### Introduce the problem

The documentation for LVGL's indev API uses functions that don't exist in version 9.1 and it is unclear how one is supposed to register these devices. Example of the issue can be found in the overview page: [LVGL Overview - Getting Started](https://docs.lvgl.io/9.1/get-started/quick-overview.html#add-lvgl-into-your-project)

Where we can see that the documentation suggests we use `lv_indev_drv_init(&indev_drv); `
But when that function is used, we get the following compilation error: `error: 'lv_indev_drv_init' was not declared in this scope`

### Proposal

Update the documentation so that it outlines how to register input devices in version 9.1 of lvgl, as the examples provided results in a compilation error.

kisvegabor, 2024-05-26:
Thank you for pointing it out! I've fixed it in #6282.

Gisbert12843, 2024-05-27:
Could we get a full pdf of the 9.1 documentation or isn't it just not ready yet?

kisvegabor, 2024-06-04:
Unfortunately, the PDF build is broken now :slightly_smiling_face:

kisvegabor, 2024-06-04:
@kdschlosser do you know what the current blocking issue is?

closed 2024-06-30 by lvgl-bot