lvgl/lvgl #6293: LVGL Integration strategy for Zephyr

0xFarahFl, 2024-05-28:
### Introduce the problem

Hi,
What is the strategy for integrating LVGL to Zephyr for every release? I noticed issues happening with the v8 (currently in use with Zephyr) and the LVGL samples, with PXP activated, won't work out of the box. 

- missing header file that should be in the same directory [(lvgl_support.h)](https://github.com/zephyrproject-rtos/zephyr/issues/71760)
- IRQ handling for Zephyr
- ...

I submitted a [PR](https://github.com/zephyrproject-rtos/lvgl/pull/52) starting to address the IRQ issue but it's getting upstreamed to v9. 
I tried LVGL  v9 as it is but it doesn't work out of the box for the existing samples. 
- Kconfig that no longer exist
[image]
- no support for Zephyr in the LVGL Kconfig
[image]

How can we use the newest release in Zephyr ?This makes it difficult for users to use PxP accelerator or for developers to contribute in general.

faxe1008, 2024-05-29:
Hey @0xFarahFl

There are in fact two pull requests pending which I opened on your behalf in upstream regarding PXP. One targeting main and therefore the next 9.X release and one for 8.4.
The one for 8.4 is waiting on another component (removal of the intermediate buffer for PXP): [https://github.com/lvgl/lvgl/pull/6158](https://github.com/lvgl/lvgl/pull/6158)
And the one for 9.0 was merged but got reverted due to missing pieces for FreeRTOS support. I reopned it again over here: [https://github.com/lvgl/lvgl/pull/6298](https://github.com/lvgl/lvgl/pull/6298)

Regarding v9 support, I have this pull request pending [https://github.com/zephyrproject-rtos/zephyr/pull/68168](https://github.com/zephyrproject-rtos/zephyr/pull/68168) feel free to test this as well.
The major blocker for landing this PR are:
- Support for a monochrome pixel format within LVGL blending/drawing (I1)
- Support for the DMA2D engine

(The latter one might be removed if we are ok with some perfomance regression on STM32 boards).

In general for the LVGL integration to work there is some gluecode needed. In the past it was decided to leave this intree within the main repository for now since it gives the possibility to fix things on our end especially in the beginning phase when a lot of things moved around in the gluecode due to new indev support and the like. Additionally things like the OSAL support rely on experimental Zephyr features (dynamic thread creation) that are still marked as experimental. Upstreaming these things right away has the high probability of a lot of overhead in case we break the Zephyr API internally. The PXP support is as small exception since there are no simple plumbing points that Zephyr can hook into nicely (at least for v8.4 that is).

So from my perspective I try to get necessary improvements fixees into upstream/v8.X and track them on our until the blockers for v9 are resolved.

0xFarahFl, 2024-06-03:
Hello @faxe1008, 

Thanks a lot for your extensive answer. I was wondering how I could contribute if I ever find out a bogus behavior? Can I make a commit that you can integrate in the PR, create a new PR depending on yours and wait for its merge or should my PRs get transferred to the lvgl/lvgl by a contributor like last time? (3rd option isn't very sustainable, I think though)

faxe1008, 2024-06-03:
> Thanks a lot for your extensive answer. I was wondering how I could contribute if I ever find out a bogus behavior?

If you can pin point the issue whether it's within LVGL or the Zephyr specific code you would either raise the issue on LVGL directly or Zephyr repository accordingly. Same goes for the PRs if you have a fix already handy.

If you are not sure where the issue is located, you can always open a issue on Zephyr repository or reach out via discord to triage the issue.

> and wait for its merge or should my PRs get transferred to the lvgl/lvgl by a contributor like last time? (3rd option isn't very sustainable, I think though)

Feel free to submit patches on the LVGL upstream and mention me as reviewer if it has implications on the Zephyr integration :^). I did just reraise your PR so it does not get lost. 

For any of the existing PR that I have open, you can submit PRs against my module fork and ping me to check them.

kisvegabor, 2024-06-21:
> Support for the DMA2D engine

@ymh-kotaro-terada and @MarkIoanAmat added support for Nema GPU in v8.4 they are working on support for v9. 
Although it's not about DMA2D but higher end GPUs found e.g. in STM32U5.

closed 2024-08-24 by lvgl-bot