lvgl/lvgl #6294: Better sync between LVGL and Zephyr for runtime hw rotation

Snevzor, 2024-05-28:
### Introduce the problem

I've made a [PR @ Zephyr project to implement display orientation setting of the ssd16xx displays](https://github.com/zephyrproject-rtos/zephyr/pull/73360).
[The display capabilities have VTILED only when orientation is 0 or 180 degrees.](https://github.com/zephyrproject-rtos/zephyr/pull/73360/commits/0afee9ec68b84fd66ae73b80d441f8dadaba228c#diff-c7011ec5b8cf3667759705c6494b80e4c9665ae7d657c427177b9971540414c5R600)
[These capabilities are retrieved in the LVGL glue code during initialization.](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/lvgl/lvgl.c#L223).
[I also added a commit for the Zephyr LVGL module to take the initial orientation into account.](https://github.com/zephyrproject-rtos/zephyr/pull/73360/commits/ce38351529a3001b6060dadc7e614c28a51a1a3f)

So far so good for setting the orientation at initialization.
For changing the orientation at runtime, there seems to be no way to communicate that SCREEN_INFO_MONO_VTILED is either added or removed. This is very specific to the sdd16xx Zephyr driver, of course. I can imagine that other display controller drivers might have other parameters that change when rotating by 90 degrees.

In a sample app I need to call:

- display_set_orientation(display_dev, DISPLAY_ORIENTATION_ROTATED_180);  --> necessary for HW rotation
- lv_disp_set_rotation(NULL, DISPLAY_ORIENTATION_ROTATED_180); --> necessary for LVGL to be aware and to switch X/Y

So 2 similar calls are needed, and currently it only works for changing orientation between 0<->180 and 90<->270 due to the specific VTILED thing mentioned above.

### Proposal

I don't know what's the best solution for this.
Perhaps if there was a possibility for a cb that is fired from calling lv_disp_set_rotation(), then it would allow for:

1. calling display_set_orientation in the glue code instead of application layer, so only 1 LVGL call remains, and the glue code manage calling the display driver through its API
2. getting the display capabilities again (which may be different for 0/180 versus 90/270 orientation).

After some googling, I found that such callback has been considered before: https://github.com/lvgl/lvgl/issues/2046
It was probably overshadowed by lots of other great improvements regarding 'rotation' that were added at that time.

kisvegabor, 2024-06-04:
Hi,

When you call `lv_display_set_rotation` `LV_EVENT_RESOLUTION_CHANGED` is sent to the `display`. Can you use it to do some hardware specific actions? See [this](https://github.com/lvgl/lvgl/blob/master/src/drivers/display/lcd/lv_lcd_generic_mipi.c#L289) as an example.

Snevzor, 2024-06-05:
Thanks for your input.

kisvegabor, 2024-06-28:
As the issue seems to be solved I close it.

If it's still not working please reopen the issue and comment here.

closed 2024-06-28 by kisvegabor