lvgl/lvgl #6296: Setting system time under Linux affects ticks

cproape, 2024-05-29:
### LVGL version

any

### What happened?

I found that the Linux framebuffer routine uses the **gettimeofday** function to get ticks. When setting the system time, it will affect the ticks error, for example: **LV_LABEL_LONG_SCROLL** no longer scrolls
I tried to implement ticks using the **clock_gettime** function and found no problems, but I'm not sure if this is correct

Here is how I implement **custom_tick_get** using the **clock_gettime** function
```c
uint32_t custom_tick_get(void)
{
    struct timespec period;
    clock_gettime(CLOCK_MONOTONIC, &period);
    uint64_t ticks_ms = period.tv_sec * 1000 + period.tv_nsec / 1000000;

    static uint64_t start_ms = 0;
    if (start_ms == 0)
    {
        start_ms = ticks_ms;
    }

    uint32_t time_ms = ticks_ms - start_ms;
    return time_ms;
}
```

### How to reproduce?

This appears when you set the system time.

liamHowatt, 2024-06-01:
True! Thank you. That implementation looks excellent to me. Do you feel like opening a PR for it?

kisvegabor, 2024-06-21:
It should be resolved by https://github.com/lvgl/lvgl/pull/6306

Please comment here if it's still not working.

closed 2024-06-21 by kisvegabor