lvgl/lvgl #6312: 8.4/9.1 Hardware based display rotation input devices

faxe1008, 2024-05-31:
### LVGL version

v9.1, v8.4

### What happened?

The code that takes the input devices coordinates and translates them to the display rotation for any given rotation looks like it contains an error (Nearly identical on v8.4 and v9.1):
[https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L639](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev.c#L639)

```
    lv_disp_t * disp = i->driver->disp;
    /*Save the raw points so they can be used again in _lv_indev_read*/
    i->proc.types.pointer.last_raw_point.x = data->point.x;
    i->proc.types.pointer.last_raw_point.y = data->point.y;

    if(disp->driver->rotated == LV_DISP_ROT_180 || disp->driver->rotated == LV_DISP_ROT_270) {
        data->point.x = disp->driver->hor_res - data->point.x - 1;
        data->point.y = disp->driver->ver_res - data->point.y - 1;
    }
    if(disp->driver->rotated == LV_DISP_ROT_90 || disp->driver->rotated == LV_DISP_ROT_270) {
        lv_coord_t tmp = data->point.y;
        data->point.y = data->point.x;
        data->point.x = disp->driver->ver_res - tmp - 1;
    }
```
When rotating by 180 degress why is the x coordinate inverted? Also: Why do you subtract 1 from the coordinate?

kisvegabor, 2024-06-07:
If e.g. `hor_res == 320` the valid range of coordinates is `0..319`. So when `x == 0`, it shall be `320 - 1 - 0 = 319`.

faxe1008, 2024-06-08:
For the making sure that the point is within the valid range this makes sense, but not for the case where the rotated points still live in the bounds of the display.

Example:
```
display_width = 320
display_height = 240

x = 25
y = 45

will yield the point (194, 25)
```
This looks odd to me. Also if you visualize this with a smaller example:
[image]
 and you rotate the display by 90° you should get:
[image]
Which is not the case with the above formula, it will yield `(1,2)` instead.

kdschlosser, 2024-06-09:
It is easier to understand like this...

[image]

and rotated 90°

[image]

kdschlosser, 2024-06-09:
if we have this

[image]

which is width = 6 and height = 8

and the target pixel is 3, 2

and we rotate it 180 degrees

x = width - x - 1 = ?
y =height - y - 1 = ?

x = 6 - 3 - 1 = 2
y = 8 - 2 - 1 = 5

you get 2, 5 for the new pixel using the math above

now check this out

[image]

kisvegabor, 2024-06-10:
@kdschlosser's example confirms me that the current calculation is correct.

closed 2024-06-17 by faxe1008